winder/Universal-G-Code-Sender #2024: UGS unable to exit jog or run mode

jasonw85, 2022-10-30:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

When using UGS I am unable to exit jog mode or run mode.  Once I jog the machine it will continue to stay in the jog mode making other tasks impossible such as using the probe tool.  I can use the probe tool and the jog but not the other way around.  I'm not sure if this is an error or I'm just missing the most basic aspect of using this software.   

### Operating System

Windows 11 home 22000.1098

jasonw85, 2022-10-30:
Found existing thread with this issue

closed 2022-10-30 by jasonw85

breiler, 2022-10-30:
I guess you found this: #1997

Can you post on that issue if that helped to resolve the problem?