winder/Universal-G-Code-Sender #2027: GCS 2.0.12 crashes after 390 seconds on Debian 11 x86_64

richardklingler, 2022-10-30:
### Version

2.0.12

### Hardware / Firmware

Grbl_ESP32

### What happened

GCS 2.0.12 crashes after 6 minutes and 30 seconds on Debian x86_64 machine without doing anything.

### How to reproduce

Start GCS 2.0.12 from console under Debian 11 on x86_64 machine and just wait.

### Operating System

Linux debian 5.10.0-19-amd64 #1 SMP Debian 5.10.149-2 (2022-10-21) x86_64

### Anything else

Every time I just launch UGS.

richardklingler, 2022-10-30:
[hs_err_pid75082.log](https://github.com/winder/Universal-G-Code-Sender/files/9896018/hs_err_pid75082.log)

breiler, 2022-10-31:
Have a look here:
https://github.com/winder/Universal-G-Code-Sender/issues/1607

I suspect that this is because either a graphics card compatibility issue or that you have multiple graphics cards on your computer.

richardklingler, 2022-10-31:
Okay..fixes it...though don't understand why a Linux setup with onboard Intel GPU is left out...as this would provide a cost effective solution for a CNC machine...you don't want to buy an expensive NVidia card for this (o;

closed 2022-10-31 by richardklingler

breiler, 2022-10-31:
This is not intentional. We don't use any advanced or new features in the OpenGL-specification but for some reason the library we are using is incompatible with some setups. The library hasn't had any new releases since 2015 and is currently the only option for us. There has been some development on it has not yet been released... =(

richardklingler, 2022-11-04:
Yepp..can confirm it works perfectly on my Debian 11 with Intel HD GPU :-)

Many thanks (o;