winder/Universal-G-Code-Sender #2033: Error while starting file stream

tobiaswuerth, 2022-11-01:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

I have a small .gcode file with some commands I'm experimenting with right now to get things right. I can run the program and it executes fine, but UGS doesn't recognize that the file is done execution. When I try to run it again I get the "Error while starting file stream" error message popup. I then have to soft-reset before I can run it again.

I noticed that the "estimated time" and "rows left in file" don't stop/update, I'm not sure if this is normal. But the state is correct in "idle".
[image]

Also, there are a lot of NullPointerExceptions in the log file [messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9912090/messages.log), like: <details>
java.lang.NullPointerException
	at com.willwinder.ugs.nbp.core.windows.DiagnosticsTopComponent.refreshValues(DiagnosticsTopComponent.java:148)
	at com.willwinder.ugs.nbp.core.windows.DiagnosticsTopComponent.UGSEvent(DiagnosticsTopComponent.java:175)
	at com.willwinder.universalgcodesender.model.GUIBackend.lambda$sendUGSEvent$3(GUIBackend.java:859)
	at java.base/java.util.ArrayList.forEach(Unknown Source)
	at java.base/java.util.Collections$SynchronizedCollection.forEach(Unknown Source)
	at com.willwinder.universalgcodesender.model.GUIBackend.sendUGSEvent(GUIBackend.java:859)
	at com.willwinder.universalgcodesender.model.GUIBackend.statusStringListener(GUIBackend.java:719)
	at com.willwinder.universalgcodesender.AbstractController.dispatchStatusString(AbstractController.java:815)
	at com.willwinder.universalgcodesender.GrblController.setControllerState(GrblController.java:704)
	at com.willwinder.universalgcodesender.AbstractController.openCommPort(AbstractController.java:346)
	at com.willwinder.universalgcodesender.model.GUIBackend.openCommConnection(GUIBackend.java:811)
	at com.willwinder.universalgcodesender.model.GUIBackend.connect(GUIBackend.java:173)
	at com.willwinder.ugs.nbp.core.actions.ConnectDisconnectAction.lambda$connect$0(ConnectDisconnectAction.java:126)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
</details>

### How to reproduce

- Open UGS
- Connect to Arduino UNO via COM4 115200
- Unlock
- Home
- Open .gcode test-file
- Execute file
- (wait till done)
- Execute again -> Error popup

My testfile:
```gcode
; starting program
G21 ; use milimeters
G90 ; absolute coordinates
G17 ;XY
G1 X10Y10 F2000
G1 Z10 F500
G18 ;XZ
G2 X25Z0 R15 F900
G2 X30Z10 R15 F500
G17
G1 X10Y10 F2000
M2
; done
```

### Operating System

Windows 11 Home Version 22H2

### Anything else

- happens every time
- tried nightly build and normal
- tried x64 and x32 build

Also, I have another issue but I'm not sure if this is related to this. I might open another issue for this but just to let you know. Sometimes when I execute a normal/larger .gcode file, it can happen that the execution stops randomly without being done or showing any error (at line 3000, 6000 or 40000 ish - doesn't matter, no pattern). I then have to soft-reset, figure out until where it run, manually remove the already executed .gcode lines and then continue. Really annoying because I can't trust my build to finish a job, so I always have to supervise it :/. I tried to increase the RAM for UGS by changing -Xms64m to Xms512m and that seemed to have helped, but the issue returned on another run, so I'm not confident afterall. I also tried changing up the gcode parser by toggling pluggins (Options > UGS > Controller Options) on or off. I also tried verbose logging without gaining any additional insights, single step mode and enabling the "echo mode" in the GRBL firmware (but doesn't show any additional information). I also made sure that all my cables are good and steady. Also checked if the Windows driver for COM Port is okay. Nothing helped so far. I have an original Arduino UNO from a Starter Kit I bought and a Protoneer CNC Shield v3 with DRV8825 drivers.

breiler, 2022-11-01:
Thanks for the detailed report!

I can't see in your screenshot if you have the "UGS Diagnostics" window open. But the log message that you refer to originates from it. If it is open can you try closing it and restarting UGS and try again. 

Regarding the intermittent disconnects sounds suspiciously like EMI-problems and I don't think that these two problems are related. It is really difficult to give advice on this as it depends on your setup. I would probably start by disconnecting the limit switches as its cabling are susceptible to EMI, plus I can't see any filtering components other than capacitors on the Protoneer shield. If that fixes the problem you probably need some filtering electronics on your inputs. If it doesn't fix the problem move on to the next peripheral such as the spindle.

breiler, 2022-11-01:
This looks really suspicious:
[image]

Too me this suggests a bad USB cable as the communication is garbled. It looks like the controller doesn't `ok` the first four lines. Make sure you have a USB cable with proper ferrite beads on it. If the problem persists, what happens if you disconnect the shield and only run the Arduino without anything else connected to it.

Regarding the EMI. Yes if there is enough current being transferred to the fairly sensitive inputs of the Arduino it can crash the CPU and you have no way of detecting it with software. This is hard problem to debug and you are not alone, read about others having similar problems here: https://groups.google.com/g/universal-gcode-sender/search?q=EMI

tobiaswuerth, 2022-11-02:
@breiler  Hmm you're right, I noticed that too. I assumed this might occur due to multi-threading or something.

I have a 1.8m **cable** without ferrite beads (had to google what those were). It's just a normal cable without those beads. I received a different cable from the starter kit ([see this photo](https://user-images.githubusercontent.com/21124705/199518936-b44d4ffe-3913-40de-84f5-4a9b9f3fecbc.png)), which seems to be shielded (?) due to the metal fabric inside of the cable but also no beads. I tried to run the script which crashes with both cables but both froze. ❌ 
I then tried to remove all **endstops** again and ran it again - without success, still froze. ❌ 
I then tried to first remove the Z **motor and driver** from the CNC shield and ran it again - froze. ❌ 
I then removed the Y motor and driver - again, froze. ❌ 
I then removed also the X/A motors and drivers - again, froze after the usual 5-6k lines. ❌ 
I then tried to remove the **CNC shield** from the Arduino and let the script run - it still froze! ❌ 

**Now the only two components left are 1) the Arduino and 2) the cable.** 👈 
I guess you could be right about the cable.

Besides that: Is there a way how I can verify that my Arduino is not the issue after all? The starter kit I bought was not from an official source. I can't guarantee that it is an official Arduino.

Additionally I remembered that during my initial testing phase I made a mistake with the driver (or it was DOA) and suddenly things started to smell funky, upon which I disconnected the power from the Arduino and the CNC Shield. One motor and one stepper driver (heatsink) were really hot to the touch. I also wasn't able to adjust the potentiometer from the stepper driver anymore. I ordered a new stepper driver and replaced the damaged one. Things worked ever since, but now that these issues occur I'm afraid that that incident might've damaged the CNC Shield or the Arduino. Is that a possibility? How can I check for something like that? ❔

Finally, I noticed some more exceptions while the script ran, like:  <details>
SEVERE [com.willwinder.universalgcodesender.GrblController]: Error while processing response <<Run|MPos:,76.747,1.995|F:2000>>: empty String
java.lang.NumberFormatException: empty String
	at java.base/jdk.internal.math.FloatingDecimal.readJavaFormatString(Unknown Source)
	at java.base/jdk.internal.math.FloatingDecimal.parseDouble(Unknown Source)
	at java.base/java.lang.Double.parseDouble(Unknown Source)
	at com.willwinder.universalgcodesender.GrblUtils.getPositionFromStatusString(GrblUtils.java:558)
	at com.willwinder.universalgcodesender.GrblUtils.getStatusFromStatusStringV1(GrblUtils.java:388)
	at com.willwinder.universalgcodesender.GrblUtils.getStatusFromStatusString(GrblUtils.java:330)
	at com.willwinder.universalgcodesender.GrblController.handleStatusString(GrblController.java:623)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:242)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:786)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:175)
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:197)
	at java.base/java.lang.Thread.run(Unknown Source)
</details>

Also, since you haven't commented on the "Error while starting file stream" I assume you assume that the issue is connected to the bad connection? Which might also be related to the  freezing issue? ❔

Thank you for your inputs!

breiler, 2022-11-02:
The `<<Run|MPos:,76.747,1.995|F:2000>>` is definitely a sign of a bad connection (or broken Arduino) and I believe that this is the reason behind the "Error while starting file stream". 

And since you have tried disconnecting practically everything from the Arduino and tried running the gcode program using two cables suggests your Arduino is toast. You don't have anything left that can cause EMI enough to cause these kinds of problems.

tobiaswuerth, 2022-11-02:
@breiler You're my hero for answering so quickly and well informed! 😃 You're much appreciated!

Before I order a new Arduino I have a few last questions I'd like to get your oppinion about.

1) I currently have to connect via a [Microsoft Travel Hub](https://www.microsoft.com/en-us/d/microsoft-usb-c-travel-hub/8nlwz0mqk26d?activetab=pivot:overviewtab), might that be an issue? 

2) Could it be that some of these settings are the issue? 
[image]
I had to adjust the steps/mm to match my 1/32 microstepping. There was no other way to get the real distance other than setting these values like they are. Since this is "only" a warning and not an error, I ignored it, because the things I tested seemed to work as expected. I read somewhere that the issue might be the CPU of the Arduino which can't handle too high frequency, but when I google the Arduino Uno it tells me that it runsArduino Uno a ATmega328P processor with 16 MHz frquency, which is a lot more than 36 kHz. Tbh I'm not sure how this all works so I was hoping you'd have some insight into these matters.

breiler, 2022-11-02:
1. I don't think that matters much that you are using a hub, you don't have any way of connecting without it to rule that out?
2. If you set $111=20000 it should no longer display the warning (you beat me how to calculate it).  But that didn't get rid of the error message or it didn't solve the issue?

tobiaswuerth, 2022-11-02:
@breiler Sadly no, my Surface only has 2 USB-C Ports and I don't have any such cables to connect directly to the Arduino.

Yes, that's exactly what I did :) And no, sadly the Error message (<<Run|MPos:,76.747,1.995|F:2000>>: empty String) and the freeze still occur. 

I noticed now that it realiably fails at the same location since I removed everything but the Arduino itself. I tried to isolate the gcode +- few lines and execute it isolated - successfully. So I'm uncertain that it is the gcode that is the issue in itself. It seems to be a "timing" or procedural issue which occurs only after executing all of the code before it. Can I ask you to give it a try and tell me if it works? [gcode.txt](https://github.com/winder/Universal-G-Code-Sender/files/9922279/gcode.txt)

UGS tells me that it "sent 6028 rows" (it's always 6028 before freezing now). The last line that I get an echo from is `[echo: G1X148.7236Y138.3695]` which I located at line 6033 in the gcode file.

breiler, 2022-11-03:
I ran the file completely without problems. After 40k lines I increased the feed speed to 200%.

tobiaswuerth, 2022-11-03:
@breiler Thank you for testing this!

I can't figure out what the issue is, so I ordered a new Arduino UNO Rev3 and hope that the issue will resolve itself. I tried to do some flash memory and RAM tests on the Arduino I already have beforehand to figure out what might be the issue. I can't say conclusivly that it is not working, but I experienced some weird phenomena when fiddling with it, so that might actually be reasonable to assume that something broke there. 

I'll keep you posted once I get the new hardware.

tobiaswuerth, 2022-11-08:
Okay, so I received my **new Arduino UNO Rev3** from the official vendor. It looks quite different to the one I had. Here are some before/after photos of the two Arduinos:

before             |  after
:-------------------------:|:-------------------------:
[image]  |  [image]
[image]  |  [image]

I successfully tested the 'hello world' "Blink" example of the Arduino - meaning the Arduino is functional. ✅ 
I then uploaded my GRBL 1.1 firmware to the Arduino (CNC Shield still disconnected from the Arduino, so just the Arduino standalone running connected to my Surface via USB Cable). Upload worked and I was able to establish a connection from UGS. I then tried to run my gcode testfile - it also worked! ✅ So that's already a win compared to the one I had before where all the exception occurred.

This fixed the issue with the "Error while starting file stream". The console and state correctly represented the actual state of GRBL and upon finishing a task UGS would recognize that the job is done and allowed me to re-send it once again without the error popup. ✅ 
Also, no "[log fragments](https://github.com/winder/Universal-G-Code-Sender/issues/2033#issuecomment-1298897017)" occurred.  ✅ 

---

So I started to test again with the CNC Shield attached to the Arduino.

Starting off everything seemed to work as expected, but then suddenly UGS became unresponsive and I wasn't able to connect to the Arduino again after restarting it. Re-connecting and thereby restarting the Arduino allowed me once again to connect UGS to the Arduino. But weirdly enough nothing seemed to work anymore. Trying to send `G1` commands didn't respond and Homing the machine just did nothing, except starting to draw power (for breaking?). What I noticed was a weird probe state was showing, unlike the [familiar numbers](https://user-images.githubusercontent.com/21124705/200581279-43178ea2-b793-4a60-8ddc-703780fa295a.png). The numbers shown are also visible when running `$#` which returns an entry for `PRB` (i.e. probe). In the response there was `[PRB:2147483.648,2147483.648,2147483.648:0]` (3x `2147483.648`, i.e. max `Int32`). I have no idea how this has been set but it stopped everything from working, even with the CNC shield disconnected again. ❌
So I started to look for a solution to whipe all the data on the Arduino and re-write the GRBL firmware. Because just 'uploading' GRBL again didn't seem to get rid of this number. I'm not sure where those states are stored on the Arduino to persist between firmware uploads. Unless something didn't work with my upload before, I don't know. Anyhow, I tried to overwrite and test the memory again to see if I just broke my Arduino again by attaching the CNC shield, as tested [in this video](https://youtu.be/McK2PSUxjck) using [this code](https://raw.githubusercontent.com/mudmin/AnotherMaker/master/arduino-eeprom-torture/torture.ino). The script provided a positive feedback, meaning that no errors where detected and all of the EEPROM has been overwritten in the process of testing. Now I was able to, again, upload GRBL to my Arduino and connect UGS to it without showing those weird coordinates. ✅ 

I then tried to connect the CNC shield again let the program run, but it crashed agin somewhere randomly. Although this time, UGS at least correctly noticed that something happened, ended the program and changed its state with the console message "connection closed". ❌
So I was again searching for the issue of what might cause my issued running my machine. I then came upon [this video](https://youtu.be/JhsJJ_cmKS4?t=312) which made me aware of a potential short circuit issue using the CNC shield I also use. Apparently the way the CNC Shield lays on top of the Arduino can potentially connect the CNC Shield to the USB Port of the Arduino. I therefor tried to block the connection by [sticking a piece of isolation tape inbetween](https://user-images.githubusercontent.com/21124705/200584848-d55dde4a-96f9-4f4a-91a2-afc665059240.png). At first it seemed to have helped it run smoothly, but then it crashed again. I left the tape there anyway, who knows. ❌
Continuing with my testing I again tried to run it without the CNC Shield, which still worked - thankfully, I was worried that the issue might be the CNC Shield and I didn't want to break another Arduino (if that was my fault somehow). I then tried to test the different motors/drivers one-by-one. First, only running the X-Axis seemed to have worked. I then tried to add Y and Z, but it didn't work. Removing Z also crashed. Removing Y again made it work again. So I tried to just run Y, which also resulted in the Arduino resetting, UGS closing the connection and obviously the script halting.  So I started to assume that the issue might be the Y-Axis stepper motor driver socket on the CNC shield or the motor driver itself. I then tried to use a different stepper driver and suddenly it didn't crash anymore. I then tried to reverse it back to the stepper driver which apparently didn't work but it suddenly did too. - Now I was confused. I haven't done anything else except maybe pressing down on the microstepping jumpers. Everything else stayed the same, but now it worked. Weird. I reattached the motors and drivers for the Z-Axis and tested again - still working, great. I then attached all the motor drivers (including X and A) and let it run again - successfully! I have let it run for the last 30min or so (2x the whole gcode test file I sent) and it hasn't crashed yet. [... cut ...] ary but all the things combined now seem to work.
---

I hope it stays that way!

If not, I'll probably order a new USB cable to connect to my Surface without the hub, including the brades. Next I'd buy the newest Protoneer CNC Shield to make sure that it is not the issue. Then if it still occurrs I'd have to look into the cables again, although I have no idea what I would have to do to guarantee that EMI is not the issue afterall.

I hope someone finds this useful.

Special Thanks to @breiler !