winder/Universal-G-Code-Sender #2036: UGS not opening on Mac

nikhiltur, 2022-11-04:
### Version

2.0.12

### Hardware / Firmware

GrblHAL

### What happened

Hello, I have been using UGS for a little while with my GrblHAL board. It was working fine a couple weeks ago, and then when I tried to open it earlier this evening it would load up all the modules and everything but no actual window would open.

I am on an M2 Macbook Air running UGS 2.0.12, and I have Java 8 Update 341, as well as Java 17 and Java 19 JDKs. I am not sure why it is not working anymore, as it was working fine a few weeks ago and I was able to use all its features properly. 

I have attached the messages.log file below as well. 
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9934204/messages.log)

### Operating System

MacOS Ventura 13.0

breiler, 2022-11-04:
We have had reports that UGS doesn't work on Mac OSX Ventura, but you mean that it has worked for you before?

How are you starting the application? Are you using the icon or are you starting it from the terminal?

nikhiltur, 2022-11-04:
Now that I realize it, I just recently updated to Ventura. I was running it on Monterey when it was working properly, only after I updated did it stop. I am using the icon to start, not using the Terminal. Would that make a difference since I am on Ventura?

breiler, 2022-11-04:
Well I saw something in the logs that it tries to use the root (/) directory for storing cache files. This problem exists on Linux and also if you run the appllication from the terminal on MacOSX. But apparently this now also happens by default on Ventura...

I would like to try removing two lines from the configuration file to see if it helps
* Open `/Applications/Universal\ Gcode\ Platform.app/Contents/Resources/ugsplatform/etc/ugsplatform.conf``
* Comment out or remove these lines: 
```
default_userdir="${HOME}/.${APPNAME}/2.0-SNAPSHOT/dev"
default_mac_userdir="${HOME}/Library/Application Support/${APPNAME}/2.0-SNAPSHOT/dev"
```

nikhiltur, 2022-11-04:
[image]

Clicking OK just doesn't do anything. I went into the config file through Terminal Nano and removed the two lines you said. I also tried downloading the Universal zip without bundled java and running from Terminal and it threw a bunch of errors. I can send that too if you want.

nikhiltur, 2022-11-04:
```
Java HotSpot(TM) 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
Nov 04, 2022 12:10:16 AM org.netbeans.ProxyURLStreamHandlerFactory register
SEVERE: No way to find original stream handler for jar protocol
java.lang.reflect.InaccessibleObjectException: Unable to make field transient java.net.URLStreamHandler java.net.URL.handler accessible: module java.base does not "opens java.net" to unnamed module @6ced7109
	at java.base/java.lang.reflect.AccessibleObject.throwInaccessibleObjectException(AccessibleObject.java:387)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:363)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:311)
	at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:180)
	at java.base/java.lang.reflect.Field.setAccessible(Field.java:174)
	at org.netbeans.ProxyURLStreamHandlerFactory.register(ProxyURLStreamHandlerFactory.java:59)
	at org.netbeans.JarClassLoader.<clinit>(JarClassLoader.java:117)
	at org.netbeans.MainImpl.execute(MainImpl.java:153)
	at org.netbeans.MainImpl.main(MainImpl.java:60)
	at org.netbeans.Main.main(Main.java:58)

java.lang.UnsupportedOperationException: The Security Manager is deprecated and will be removed in a future release
	at java.base/java.lang.System.setSecurityManager(System.java:425)
	at org.netbeans.TopSecurityManager.install(TopSecurityManager.java:525)
	at org.netbeans.core.NbLifecycleManager.advancePolicy(NbLifecycleManager.java:69)
	at org.netbeans.core.GuiRunLevel.run(GuiRunLevel.java:84)
	at org.netbeans.core.startup.Main.start(Main.java:295)
	at org.netbeans.core.startup.TopThreadGroup.run(TopThreadGroup.java:98)
	at java.base/java.lang.Thread.run(Thread.java:1589)
```

breiler, 2022-11-04:
No, I expect that the zip-bundle logs will only contain error logs regarding a incompatible java version, so those logs will not be helpful.

I'll see if I can find a working combination for the settings, hold on. 😄

nikhiltur, 2022-11-04:
Sure, take your time, thanks so much for your help :)

breiler, 2022-11-04:
Add these two line in the beginning of the configuration file:
```
default_userdir="${DEFAULT_USERDIR_ROOT}"
default_cachedir=${DEFAULT_USERDIR_ROOT}/var/cache
```

If it still doesn't start I would like to see the log file, thanks!

nikhiltur, 2022-11-04:
Hey I am not sure what happened, I put those two lines that I originally removed back in and it opened once. I quit the app and tried reopening it and it did not show up again. Just wanted to keep you updated.

nikhiltur, 2022-11-04:
> Add these two line in the beginning of the configuration file:
> 
> ```
> default_userdir="${DEFAULT_USERDIR_ROOT}"
> default_cachedir=${DEFAULT_USERDIR_ROOT}/var/cache
> ```
> 
> If it still doesn't start I would like to see the log file, thanks!

Should I replace the two removed lines with this?

breiler, 2022-11-04:
Yes please

nikhiltur, 2022-11-04:
The first time I opened it after changing the config, it opened properly. I quit the app and reopened again (since the same thing happened last time I changed the config file), and it has since stopped opening (it shows loading modules as done but no visible window opens, same issue as the beginning). I have attached the log file below.
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9935328/messages.log)

breiler, 2022-11-04:
Very strange... The log doesn't give any clues.

What happens if you delete everything under `/Users/nikhilturlapati/Library/Application Support/ugsplatform/var/cache/*`?

nikhiltur, 2022-11-04:
The issue persists. Once the app has been force quit through Activity Monitor the cache files all are regenerated, except the lastModified folder, which is generated upon attempting to open the app.

breiler, 2022-11-04:
Sorry, I am out of ideas...
It is difficult for me to reproduce as my Macs are too old to run Ventura.

nikhiltur, 2022-11-04:
Aw man, that's okay. Thank you for your help. For right now I suppose I will just change the config file every time I need to open the app :'). Or I will keep it open all the time until the issue has been resolved. Thank you for your time

sdempsay, 2022-11-23:
I got it to work on my M1-Pro by getting a Java8 build from https://www.azul.com/.

protozoo, 2022-12-08:
Hi there,

attempting to run UGS 2.0.12 on a 2020 M1 Macbook Air with Ventura 13.0.1  and having the same issues described above. I had UGS installed (and working) months ago but I recently installed Ventura and when trying to run existing UGS version (installed about a year ago) I got the loading/splash screen and then nothing: no window, toolbar menus not showing dropdowns, etc... So I deleted the cache folder (actually, the whole Library/Application Support/ugsplatform folder) as suggested above and I got the UI to show up. However the USB port dropdown didn't work (it showed "cu.wlan-debug" and wouldn't react when clicked). Same behavior with other dropdowns such as Firmware and Baud. Felt like the UI is broken.

So tried manually typing the port name (tried both "cu.usbserial-11120" and "/dev/cu.usbserial-11120") and clicking the Connect button but nothing, and UI becomes even more irresponsive.

I've also tried downloading and running the latest UGS version (2.0.12) but I get the same behavior. I am attaching the logs file, where I see an error with a Java class. This error gets appended to the log when I click the Connect button but, as said, the application presented the other problems (no UI after splash screen, irresponsive dropdowns...) even before that.

Pleae find my logs file here: [messages.log](https://github.com/winder/Universal-G-Code-Sender/files/10180988/messages.log)

Many thanks for the software and for your help

protozoo, 2022-12-19:
Hi there, any chance of looking into this or getting any hint? @breiler maybe? 
Many thanks

breiler, 2022-12-19:
@protozoo, no can do as my Mac's are too old to run Ventura. 😞

protozoo, 2022-12-19:
Thanks for the reply @breiler . Anything I could do on my side to help debug it? Thx

fehguy, 2022-12-23:
I can confirm that today's nightly (`20221222`) does not have this issue on MacOS 13.0.1 (22A400) 👍

nikhiltur, 2022-12-23:
Once I get home I can download and test again as well, hopefully it works I much prefer this over anything else at the moment.

protozoo, 2022-12-23:
Happy to report that today's nightly (20221222) is working for me too (the "all platforms" version; the Mac one froze on start and didn't go beyond that). Thanks

FPOBDOC, 2023-01-27:
I had the same problem as described above.  Installed the current Mac version (2.0.13) and it works fine, including the visualizer. Glad that this has been solved...
Running Ventura 13.1 on a 2020 M1

closed 2023-01-27 by breiler

yangit, 2023-09-15:
I have the same problem as described above.
I'm on Macbook M1 Pro Ventura 13.5.2 (22G91)
UGS Version 2.0.20
I have had UGS working some time ago, and then it start to hang on splash screen.
I have checked `/Applications/Universal\ Gcode\ Platform.app/Contents/Resources/ugsplatform/etc/ugsplatform.conf`
And it contains 
```
default_userdir="${DEFAULT_USERDIR_ROOT}"
default_cachedir="${DEFAULT_CACHEDIR_ROOT}/var/cache"
```
at the top, so I did not touch it.

I got UGS to load a few times by rebooting and restarting, but then the serial port dropdown does not contain my CNC controller, which is visible on the same version of UGS on another computer.