winder/Universal-G-Code-Sender #2042: Door close button

Diddern, 2022-11-09:
### Description

When a door alarm is triggered, there is no way (that I can find) to reset the door alarm without pressing the play-button. A double press of the play-button will resume a running program, which is error-prone.

We have a [button for opening door](https://github.com/winder/Universal-G-Code-Sender/pull/1558) but a button for clearing the door alarm button should exist.

daxliniere, 2022-11-14:
I _think_ this is something related to the CNC controller that UGS is interfacing with.
What are you using?
And why do you need to double-press play? And what problem do you get when you do?

breiler, 2023-02-23:
I can't reproduce this. If I trigger the door switch it enters a HOLD-state. I can either press play (once) to continue the program or press play to stop it. I am using a vanilla GRBL 1.1f

daxliniere, 2023-02-24:
>[..] or press play to stop it.

@breiler, I think you meant 'press stop'?

breiler, 2023-02-24:
@daxliniere yes of course, I meant `press stop` 🤦