winder/Universal-G-Code-Sender #2043: Colour-code debug information in Console

daxliniere, 2022-11-09:
### Description

In FluidNC (and others?) we can enable debug messages. It would be great if they were colour-coded in the Console.
Commands include **MPos:** and **DBG:**

I can imagine an improvement to this feature (that could come later) where the user can themselves define in their settings which messages to colour and what colour they should be. This could go into the Fonts & Colors tab of UGS settings.

Perhaps this is even linked to the protocol (Firmware) that is used. (FNC, GRBL, etc)

[image]

daxliniere, 2022-11-09:
Maybe even all "ok"s can be green, too?

daxliniere, 2023-08-19:
If you can give me examples from each protocol, I'd be happy to do some leg-work assigning colours.

breiler, 2023-08-19:
NetBeans comes with a predefined color scheme which I inteded to use:
[image]