winder/Universal-G-Code-Sender #2049: The rendering of the UI under Linux is faulty

dasarne, 2022-11-14:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

When starting UGS under Linux, parts of the UI are not rendered. If you send an event to these elements (mouse over), they will be rendered. However, the objects do not remain up to date after that.
Resizing the window is blocking the UI.
[image]
[image]
[image]
[image]

### How to reproduce

- Download Version 2.0.12
- `rm -rf ~/.ugs ~/.ugsplatform `
- Extract and start UGS
- The presentation of the UI is incorrect
- Resize the window
- The UI is unreachable

### Operating System

Kernel: 5.15.76-1-MANJARO arch: x86_64 bits: 64 Desktop: KDE Plasma     v: 5.26.2 Distro: Manjaro Linux

### Anything else

These warnings can be seen:
```
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/arne/Schreibtisch/ugsplatform-linux/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```
The following messages can be found:
- [messages-1.log](https://github.com/winder/Universal-G-Code-Sender/files/10002028/messages-1.log)
- [messages.log](https://github.com/winder/Universal-G-Code-Sender/files/10002015/messages.log)

I'm using xorg-server 21.1.4-1

breiler, 2022-11-14:
Please try the latest nightly build. I updated the OpenGL libraries which should resolve this problem.

dasarne, 2022-11-14:
Thank you Joacim. 
Everything works perfectly! 

[image]

closed 2022-11-15 by breiler