winder/Universal-G-Code-Sender #2050: Bug?? Toolbox missing from snapshot 2022-11-04

daxliniere, 2022-11-14:
### Version

Other

### Hardware / Firmware

Other (add details about it below)

### What happened

[image]

### Operating System

Win 10 Pro N x64

breiler, 2022-11-15:
Can you post your log file?

breiler, 2022-11-15:
As you seem to have got it working in #2052 I assume that this is a settings-related problem. The settings problems will hopefully eventually get better as of version `2.0.13` as they will now share config directory between versions.

daxliniere, 2022-11-15:
Unfortunately not, if you check the screenshots you can see 2022-11-04

daxliniere, 2022-11-29:
Resolved in 2022-11-29 build.

closed 2022-11-29 by daxliniere