winder/Universal-G-Code-Sender #2055: Importing SVG on design sheet with wrong coordinates

dasarne, 2022-11-16:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

The svg file "[Test.svg](https://user-images.githubusercontent.com/8106470/202260360-b0b6993d-41ea-4992-b2ff-cbf55f10479e.svg)" contains a rectangle with dimensions 50x70 mm. The coordinate origin in this file is at the bottom left.
[image]
When you import this svg file into a design, it has dimensions of 270x190 mm. The origin of the coordinates is in the upper left corner.
[image]

### How to reproduce

You create a new design.
Then open the file "[Test.svg](https://user-images.githubusercontent.com/8106470/202260360-b0b6993d-41ea-4992-b2ff-cbf55f10479e.svg)". The representation of the rectangle is not correct.

### Operating System

Kernel: 5.15.76-1-MANJARO arch: x86_64 bits: 64 Desktop: KDE Plasma     v: 5.26.2 Distro: Manjaro Linux

breiler, 2022-11-16:
Thanks for the report. I found some problems with some affine transformations that doesn't calculate the size correctly. There is also a couple of problems with line objects where the properties is all zeroed out. So there is a bit to work with.

dasarne, 2022-11-16:
I tried to give you an easy example file. If you are ready, give me a hint, and I'll test it again. [image]

PR #2151 (2023-02-03): Fixes for SVG import
* Import SVG:s synchronously and applies the units to the scale transform.
* Shapes with no width or height will be taken into account in a group.

Fixes #2055
Fixes #2056
closed 2023-02-03 by breiler