winder/Universal-G-Code-Sender #2061: partial gui

grumpsshed, 2022-11-23:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

New install of debian 11 Bullseye, Fresh download of UGS and only getting a parital gui as per screenshot.

[image]

### How to reproduce

OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/root/ugsplatform-linux/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

### Operating System

Debian 11 Bullseye

### Anything else

Every time

breiler, 2022-11-23:
This is a known issue, try the latest nightly build. It has the latest OpenGL library that should resolve this problem.

grumpsshed, 2022-11-23:
Yep resolved issue, many thanks.

closed 2022-12-30 by breiler