winder/Universal-G-Code-Sender #2077: Sending to Pendant api gcodes throwing errors

MadTooler, 2022-12-09:
### Version

2.0.12

### Hardware / Firmware

GrblHAL

### What happened

While sending gcode position update commands "G10 P0 L20 X#.##Y#.##" via pendant api, main gui is generating error9 where other commands from the console may be initiated:
```>>> $J=G20G91X0.1F600
>>> G10 P0 L20 X42.07Y7.12
ok
[Error] An error was detected while sending 'G10 P0 L20 X42.07Y7.12': (error:9) G-code commands are locked out during alarm or jog state. Streaming has been paused.
**** The communicator has been paused ****
```
or error2 that seems to keep things locked up until a soft reset:
```
>>> G10 P0 L20 XX0.00Y0.00
[Error] An error was detected while sending 'G10 P0 L20 XX0.00Y0.00': (error:2) Missing the expected G-code word value or numeric value format is not valid. Streaming has been paused.
**** The communicator has been paused ****
```
It is not clear if the error2 is as common, or where it originates, as the error9 where there does not seem to be robust enough interaction between the console and the pendant api to not overlap commands.  Also to note error9 seems to allow other commands to get through and un-pause the feed while error2 keeps things stopped until the soft reset is performed.

May be related to older issue [https://github.com/winder/Universal-G-Code-Sender/issues/1043](url)

### How to reproduce

Send gcode commands through pendant api at 5 to 8hz while a gcode program is not being run.  Occasional inputs from the console, such as reset zero or flood, may cause the interrupted error9.  error2 is harder to determine where the source is with the current system.  

NodeRed is being used to send the messages to the pendant api.

### Operating System

System: 32-bit Kernel version: 5.15 Debian version: 11 (bullseye)

### Anything else

It is sporadic.  It may run fine for 10 minutes then suddenly have the error almost constant for a minute.

breiler, 2022-12-09:
In the  #1043 he is using an additional hardware that is communicating with the controller so it is a bit different than what you are trying.

Regarding the error2, this looks like a communication error where data is lost. This usually mean that there is too much electrical noise on the USB-cable. Make sure that there are no high power cables near the USB-cable. Also make sure you have a properly shielded USB-cable. I think error9 is related to the previous error.

MadTooler, 2022-12-09:
Thank you for the reply.

Note I am using the current nightly build.  

I agree error2 definitely looks like a communication error.  Where the error comes from is unclear.  My chain of systems and parts, including the pendant api itself, passing along that message is stretched.  Regarding possible electrical noise in the USB cable, noise is always a possibility.  However, I am not very convinced it is the main culprit here, at least not so much with error9.  If you are referring to the USB cable between UGS and the grblHal device, I ran numerous programs through without the pendant api injections prior without a single error notification.  I plan to add an additional check to the gcode string before it is sent to the pendant api just in case it is in that section of my system.  When I get to it, I plan to add a counter for the number, if any, bad lines it has to reject.

When looking over the console messages, error9 appears more like what the error notes : "G-code commands are locked out during alarm or jog state."  It appears, to me, the new command sent via pendant api tries to inject while grblHal is still processing the prior sent prior by the pendant or the console.  This is mostly when from the console as with the actual jog command that was sent and before the OK return the pendant shoved in the latest message which triggered the rejection.

Is there any way beyond the console to listen to what the pendant api is sending?

breiler, 2022-12-12:
Having a closer look at your console log output and the code - the ">>>" output contains the actual command being queued through the PendandAPI and sent by UGS. So that indicates that it might not be a (USB) communication error after all. I think that you should double check the code that injects commands through the pendant API.

Here is the code that prints the output in UGS:
https://github.com/winder/Universal-G-Code-Sender/blob/38cb90f19667d820bea6d8a449cfb4b2385b29f4/ugs-core/src/com/willwinder/universalgcodesender/AbstractController.java#L677

> Is there any way beyond the console to listen to what the pendant api is sending?

When sending commands through the pendant API you will add commands to a central command buffer in UGS. All commands in this buffer will be processed before processing the file stream. In other words, the output from the console (with the ">>>" lines) is a combined ordered list with all commands sent from the pendant API and file stream.

breiler, 2023-01-11:
Any updates on this?

breiler, 2023-01-11:
I was able to produce this error when streaming gcode very fast where some internal buffers got out of sync due to concurrency. So it could be related. We can keep this issue open, let me know if any new errors occur.

MadTooler, 2023-01-11:
> 

Sounds good.  Will do.

breiler, 2023-01-25:
Issue seems resolved, so I am closing this now.

closed 2023-01-25 by breiler