winder/Universal-G-Code-Sender #2078: Problem opening file: Corrupt data found while processing gcode steam: G21??G21??1??

munusshih, 2022-12-09:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

I tried to import my .gcode files into Universal Gcode Platform but everytime it gives me this this error message.
`Problem opening file: Corrupt data found while processing gcode stream: G21??G21??1??`

My gcode was in SO-2 style and I have tried many other gcode files that I found online but still couldn't find a way to solve this. Everytime I hit `send` it would also give me this error message saying that `Error while starting file stream.`

### Operating System

Windows 10

### Anything else

Everytime I import a gcode file in it. Everything else works perfectly fine, such as Return to Home position or Get State.

breiler, 2022-12-09:
Looks like a duplicate of https://github.com/winder/Universal-G-Code-Sender/issues/2011
Read through the issue and try the suggested fixes. If that doesn't help I have no idea, somehow it is running some old code even though you are using 2.0.12.

breiler, 2022-12-12:
How did it go?

You could try the latest nightly build to see if it helps.

munusshih, 2022-12-12:
Hi @breiler Thanks for the suggestions. Unfortunately I have tried permanently deleting all of the folders, clearing the caches, downloading different versions (Nightly built / 2.0.12 both x32, x64) and even reboosted the system several times but none of them seem to work......

I noticed one thing that this error message is caused by the "visualizer" function and I would not get the error message if I had the visualizer turned off. But I still wouldn't be able to send the data into my CNC without getting the error message `Error while starting file stream.`.

A little more context: I am also using Windows 10 x64 and the CNC machine I'm using is called Sovol SO-2.

breiler, 2022-12-12:
This error will block you from using UGS. 

UGS will preparse the gcode (applying preprocessors and filters) and save it to disk in a special format to be able to stream the commands quickly to the controller. This format changed a couple of months ago to fix another issue. The line `G21??G21??1??` is the old format and there is no longer any code in the version 2.0.12 that can generate this. I can't explain how this happened and I can't reproduce the error on my test system.

The best recommendation I can give you is to try the latest nightly build. We have changed where UGS will store cached files and settings so this will hopefully resolve this:
https://github.com/winder/Universal-G-Code-Sender#downloads

munusshih, 2022-12-12:
Unfortunately the nightly build did not help. The same error happened to both UGS platform and classic nightly build. Same error code with ???? in it.

This is my first time using UGS and I did not have any older version of UGS downloaded on my computer ever so it is unlikely that it’s a result of old cache.

Is it possible that the version that is uploaded here (that you claimed could never generate the error code) is not the “latest version”, that someone had uploaded a wrong file? Or one of the links is mixed up?

As there are so many versions + platforms, I’ve only downloaded them from GitHub.

Just to brainstorm some possibilities. It might not be any of that.

breiler, 2022-12-12:
I just tried downloading the latest nightly build of UGS Platform Win64 and I could load a gcode file with no problems. I also tried the UGS Classic nightly build and it too worked without problems. 

From where did you download UGS?
* https://github.com/winder/Universal-G-Code-Sender#downloads
* https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.12
* https://winder.github.io/ugs_website/download/

They all point to the same files... But just to rule that out.

If you download the win64-bit version 2.0.12, how large is the zip file? It should be 145 919 718 bytes.

breiler, 2022-12-12:
I have attempted to fix this by writing/reading file as UTF-8 instead of using the OS encoding.

It is available in the latest nightly build: 
https://github.com/winder/Universal-G-Code-Sender/releases/tag/nightly

munusshih, 2022-12-12:
Amazing!!!! It worked!!!!!
Thank you so much.

closed 2022-12-13 by breiler