winder/Universal-G-Code-Sender #2085: Arduino Due not recognized as Joypad

SailWithChips, 2022-12-20:
### Description

Same as issue #1564. I'have done my pendant using an arduino Due. The reason behind, instead of using Micro is because with Due you can implement lots of inputs, outputs and also analog inputs.

I have done what stated into issue mentioned above, download and run **"SDL2 Gamepad Tool"** and description of my joystick is: "03000000412300003e00000000000000,Arduino Due,platform:Windows,a:b10,b:b8,x:b9,y:b11,guide:b12,leftshoulder:b0,rightshoulder:b1,dpup:-a1,dpdown:+a1,dpleft:-a0,dpright:+a0,"

So please if you can be so kind to add this description to your list would be appreciated and also possibly for other users.

By the way, using SDL2 Gamepad Tool and clicking on "Set mapping as environment variable" I have already working and UGS, fromt then, detected as joystick but this is on Windows platform... I have other CNC machine with a raspberry and would be nice to have also there working so, anyway would be interesting to have this on "gamecontrollerdb.txt".

 I have developed a new PCB, very simple at this moment to be plugged directly on a due as a shield... it's very handy.

Thanks a lot for you nice job developing all this very good and helping software!

breiler, 2022-12-26:
I have added it as game controller: https://github.com/winder/Universal-G-Code-Sender/commit/99e620208dc129a5d49a2841a8647a712f630770

It should be available in the latest nightly build as soon as this build is done: https://github.com/winder/Universal-G-Code-Sender/actions/runs/3779905927

I am not sure if this will work on Linux though as the platform is specified as "Windows". Let me know if it works.

closed 2022-12-28 by breiler

SailWithChips, 2022-12-28:
Thanks!!, going to test it with rpi as soon as I can!
Hope it can be usefull for other users ;)