winder/Universal-G-Code-Sender #2094: Describe the featureSupport for Wifi Connection to MKS DLC32 v2.1+ Based boards

ArcAIN6, 2022-12-22:
### Description

I like the fact that the MKS DLC based board doesn't tether my CNC machine the PC, and for that reason, i purchased on. It works pretty decent, but the only draw back, is i like using the UGS platform, and "app" or "web based" leaves a LOT to be desired. For that reason, and the fact my CNC machine is 16 feet away from my computer, a usb cable that long seems silly... especially since the mainboard supports WIFI connections. 

Is there any plans now, or in the future to support these boards using a WIFI / IP connection instead of com port? 
I searched the "issues" section to see if there was already a post about this, but i didn't see any. If this is a double-post, or already answered kind of thing, i apologize, I'm just really keen to get this working. 

Thanks in advance 

Additional information: 
Git for the Makerbase MKS DLC32: https://github.com/makerbase-mks/MKS-DLC32
They usually cost around $15 - $20 USD on allieexpress, a little more on ebay, and amazon.

breiler, 2022-12-23:
UGS has the ability to communicate with a controller using TCP/IP, it is however a bit cumbersome to setup.

There are some instructions here: https://github.com/winder/Universal-G-Code-Sender/wiki/Connecting-the-Controller#smoothie

1. Tools -> Options.
2. Click UGS.
3. Open the Sender Options tab.
4. For Connection Driver choose TCPDriver and click OK.
5. Back in the main page, under Firmware at the top choose ~~Smoothie~~ GRBL.
6. For Port, type the IP address or hostname of your controller.
7. For Baud type the port number to connect to 23.

ArcAIN6, 2022-12-23:
oh cool Didn't see that. Works perfectly now!

closed 2022-12-23 by ArcAIN6

Gabriel28Anselmo, 2023-12-07:
olá pessoal
estou tentando rodar a minha cnc no universal g-code sender, com a plana MKS DLC32 V2.1
porem nao estou conseguindo

alguem consegue me ajudar?

breiler, 2023-12-07:
@Gabriel28Anselmo please start a new discussion and provide more details on what you need help with. 

We need to know a bit more about your setup, for instance what firmware is the board running? How are you connecting to the board (WiFi or USB)?

Robhoski, 2024-05-13:
I'd like to know how an average Joe who isn't a programmer is supposed to know how to navigate all this information? I just want to restore the factory settings that I screwed up when I tried to flash my MKS DLC32 V2.1 card for my homemade CNC table. Good lord this stuff is a nighmare. Why can't it just be a simple click on this and upload it type of situation? Why do people have to be computer gurus to understand all of this crapola - which is one link after another that propels you deeper and deeper into the rabbit hole of hell to make you see more and more folders filled with mysterious files and a zillion more options. Holy Crapoly. How can a simple carpenter and bodyman like myself find a solution without having to return to college for four years? HELP!

breiler, 2024-05-14:
@Robhoski I understand the frustration. I'd say, contact the company that sold you the hardware. Unfortunately I know that the company can be difficult to get in touch with. I have tried contacting them myself attempting to get sponsring with hardware to try and improve the user experience, with no luck.

This essentially means that the open source projects (runned by volunteers) are stuck with the support which we have to do for free. Worth keeping in mind.

Your controller came with an old version of ESP32_GRBL. I have no idea of the preferred way of restoring it.

However I do know that the controller is capable of running FluidNC (another free and open source projects led by volunteers). If you have the patience I might be able to guide you to install.

I wrote a web based installer for FluidNC here: https://installer.fluidnc.com

With that you should be able to flash the controller.

Next you'll need to create a config file which can also be done with the installer. Here is some documentation about the hardware and configuration examples:
http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32

If you need further assistance, please create a new discussion thread.

Robhoski, 2024-05-14:
I did eventually get it resolved. Someone was kind enough to turn me on to
the process of downloading the MKS laser tool to flash the board with the
appropriate bin file. Thanks.