winder/Universal-G-Code-Sender #2095: Possible issue: Illegal reflective access

daxliniere, 2022-12-24:
### Version

Other

### Hardware / Firmware

FluidNC

### What happened

I made a .bat file for launching the latest version without having t update my Windows toolbar button every time and I got this warning. Not sure if it's of any interest to you. I'll be suppressing the warning.

The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/C:/Program%20Files/DEV/UGS%202-0-13/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

### Operating System

Win 10 Pro N x64

### Anything else

Please update the UGS version dropdown menu for 2.0.13 :)

breiler, 2022-12-26:
Yes, that warning has been there forever... =)

It originates from the Netbeans platform that we are using so there isn't much we can do about it. I am closing this as won't fix.

closed 2022-12-26 by breiler

daxliniere, 2022-12-26:
👍👍