winder/Universal-G-Code-Sender #2096: Macros are not saved

NilsNippe, 2022-12-26:
### Version

2.0.13

### Hardware / Firmware

GRBL 1.1

### What happened

Macros don't get saved on restart of the application, both in x64 and x32 [?] mode.
They work as expected when writing them, but aren't available in the macro toolbox after closing and reopening the application.
They also diappear from the macro editor.
Log: [messages.log](https://github.com/winder/Universal-G-Code-Sender/files/10304953/messages.log)

Relevant log line: SEVERE [com.willwinder.universalgcodesender.utils.SettingsFactory]: Can't save settings!
java.io.FileNotFoundException: C:\Users\nilsn\.ugs\UniversalGcodeSender.json (access denied)

The file does exist and is not write protected.

### How to reproduce

1) Open UGS
2) add and save macro
3) close UGS
4) open UGS

Macro does not exist anymore.

### Operating System

Windows 11

breiler, 2022-12-27:
Do you have any software (such as a antivirus software) that might be blocking you from writing to `C:\Users\nilsn\.ugs\UniversalGcodeSender.json`?

NilsNippe, 2022-12-27:
Only Windows Defender, which doesn't show anything relating to UGS in the protection history.

breiler, 2022-12-27:
Try and open it manually with notepad and try to modify it.

I can not reproduce this error, it must be related somehow to your OS.

NilsNippe, 2022-12-27:
Opening with notepad, adding somethin and saving worked, but didn't change the behavior after cosing and reopening.

I deleted the json file altogether and opened UGS, causing it to be created again. The new json file was not hidden anymore and saving the macros were saved correctly when closing.
The same behavior can be archieved by changing the "hidden" setting for the json file.

As I'm not sure if the problem is with my machiene, I'm leaving it up to you if this issue should be closed. My problem is soved.

breiler, 2022-12-27:
I am glad it is working again. 

This is the first time I heard of the problem, but now I'm aware that it can happen on some systems and I'll keep an eye out if it happens again.

closed 2022-12-27 by breiler