winder/Universal-G-Code-Sender #2119: Visualiser and Joystick setting not being brought into new versions

marshr32, 2023-01-12:
### Version

2.0.13

### Hardware / Firmware

GRBL 1.1

### What happened

Both 2.0.13 and the latest nightly are not bringing across my visualiser and joystick settings from 2.0.12.
Window positions seem to be read OK (but only on the second startup with 2.0.13)

### Operating System

Win 10

breiler, 2023-01-12:
In version 2.0.12 and earlier we had a setting that should have placed configurations files according to this:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#configuration-files

Unfortunately this setting was wrong which sometimes led to configuration files being placed in random places. The issue is fixed as of 2.0.13 and the configuration files should be retained between upgrades. There is no way for us to automatically migrate them to the new location so they are unfortunately lost...

Closing as "won't fix".

closed 2023-01-12 by breiler

marshr32, 2023-01-12:
No worries, it's not a huge job to set them again, just thought I'd mention it.
Thanks for all your awesome work on UGS 😁

marshr32, 2023-01-13:
For anyone who sees this later, you can export settings from the old version and import them in the new one. This gets most things sorted.