winder/Universal-G-Code-Sender #2121: Run From.....  often not available

marshr32, 2023-01-12:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

I've had a couple of jobs stop 30000 odd lines into a 40000 line job and needed to use "run from..." to restart the job where it left off, but it's very often greyed out. Only shutting down UGS and reopening would give me the option to use it.
This is not ideal as it means rehoming etc and the possibility of doing something wrong and losing machine position to continue successfully.

When should the option be available?

Note this is in 2.0.12 and I haven't tried 2.0.13 or the nightly yet.

breiler, 2023-01-12:
Just to make sure. You mean the "Run from..." from the menu bar and not from the editor right?

This action from the menu bar has some logic that checks if the machine is connected and that there is a gcode file loaded:
[image]

This action in the editor does not contain such logic:
[image]

marshr32, 2023-01-12:
Yeah the one in the menu, I didn't realise there's one in the editor so that's good to know for the future.

There was a file loaded and the machine connected as it was just cutting that file when it stopped, but the run from command was greyed out after the soft reset (possibly also before, I didn't look)

marshr32, 2023-01-16:
How is the Run From... command meant to work?
I've highlighted a line in the editor and right clicked, then selected run from, but nothing happens. If I press play at this point it starts the code from the beginning.
Run From.... is greyed out again despite having just run part of this program.

What am I doing wrong?

breiler, 2023-01-16:
This is how it is supposed to work:
[image]

I have just booted up my Win10 machine and tried this with both v2.0.12 and v2.0.13 and it works as expected. The menu is never greyed out. Do you mind sharing one of your files so that I can test with them.

breiler, 2023-01-16:
> So after right clicking and selecting run from I should be able to push play and it'll run from that point?

Yes, that should work

Regarding the action getting greyed out, I can't reproduce it. But it must be some event that is not getting dispatched properly. It requires some investigation.

> Is there a way to bring up line numbers in the editor? 

They should have been enabled by default, but I guess there is a difference between OS:es. It is odd that you didn't get the toolbar either... Well you can right click the gray area left of the gcode and choose "Show line numbers".

[image]

marshr32, 2023-01-16:
OK, I've found one dumb thing I was doing wrong.

In the editor the line you want to start at needs to be left clicked to highlight it before right click>run from
I was just right clicking on the line which I guess is why it started from the start of the program, duh!