winder/Universal-G-Code-Sender #2145: UGS stops about every 10 minutes

bcnx, 2023-01-28:
### Version

2.0.13

### Hardware / Firmware

GRBL 1.1

### What happened

So my job stops every 10 minutes (more or less) and I get the message "finished sending file in 00:09:33: I can resume by hitting the Play icon. Not sure what is going on. Wil test next with the classic version. 
Thanks for your insight. I'm not sure about the GRBL version by the way, need to look for that.

### Operating System

Linux Mint

breiler, 2023-01-29:
Can you attach a sample gcode program. Is there a reason you are running GRBL 0.8, it is very old...

bcnx, 2023-01-29:
Hi,
it's the version that came with the CNC board. It's my understanding that I cannot just flash it to a higher version as it requires changes in the hardware. Might consider a whole new board though. I don't think the problem is in the gcode (attached anyway), but in the USB port being put to sleep. I disabled power saving and even killed the power saving daemon, but still the same ...

Cheers!

[Uploading mc_fire_birds2_clean.txt…]()
BC

breiler, 2023-01-31:
Something went wrong with your file upload, can you try and attach it again?

breiler, 2023-02-11:
Closing we didn't get more information

closed 2023-02-11 by breiler