winder/Universal-G-Code-Sender #2155: Internal parser error: missing data

VovoNauta, 2023-02-10:
### Version

2.0.13

### Hardware / Firmware

GRBL 0.9

### What happened

When I try to apply to Gcode after load file, generate test data or Scan surface I get the error: Internal parser error: missing data.

### How to reproduce

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/10706450/messages.log)

### Operating System

win10

### Anything else

When I try to apply to Gcode after load file, generate test data or Scan surface I get the error: Internal parser error: missing data.

breiler, 2023-08-18:
Please try with version 2.0.19, make sure to double check your settings after upgrading:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.19

breiler, 2023-10-13:
Closing as we got no further information

closed 2023-10-13 by breiler