winder/Universal-G-Code-Sender #2157: Can't build. Maven throws error.

j00ch, 2023-02-13:
### Version

2.0.14

### Hardware / Firmware

Other (add details about it below)

### What happened

I can't build this. Trying to build this on Arch Linux, but i get an error about maven:

[error.log](https://github.com/winder/Universal-G-Code-Sender/files/10717936/error.log)

Any hints?

### Operating System

Arch Linux

breiler, 2023-02-13:
What Java and Maven version are you using?

I would recommend that you use Java 11 or 13. Newer versions of Java are not compatible with the Netbeans APIs we are using. Java has also dropped the JavaScript engine in later releases.

j00ch, 2023-02-13:
I'm using Java 13.0.1 build 9 as described in the directions. I added the following line just before the build process.
`export JAVA_HOME=/usr/lib/jvm/java-13.0.1-openjdk`

PKGBUILD that I'm using has been customized from this one: [https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=universal-gcode-sender-git](url)

breiler, 2023-02-13:
Then I have no idea what is happening, I saw one issue resolving similar problem by erasing the ~/.m2 directory:
https://github.com/eclipse-californium/californium/issues/420

j00ch, 2023-02-13:
> I saw one issue resolving similar problem by erasing the ~/.m2 directory: [eclipse-californium/californium#420](https://github.com/eclipse-californium/californium/issues/420)

Wow, that was the fix. Thanks for that one, builds fine now.

closed 2023-02-13 by j00ch