winder/Universal-G-Code-Sender #2168: Add a Machine Database and multiple instances.

GrahamJB, 2023-02-27:
### Description

I have 6 routers, all Grbl based but they are not all the same, different sizes, some enable homing, some don't......

I want to use UGS for different routers, sometimes I would like to run more than one at the same time from the same PC (My PC can take it!).

First a machine database,
UGS goes through various settings in the setup wizard, if I connect to a new router then these are now wrong! So I have to re-run the Setup every time, other things like probing settings should also be saved here so I can just click on the router name and they are all now correct.

Second Multiple instances.
I can only use UGS to connect to one router at once from a single PC, so to control two routers means 2 PC's, three... it's a linear progression but unfortunately a very expensive one.

Why can't I run multiple instances of UGS on a single PC to control 2 or more routers at once?

I know this is two features but I think any solution is going to be related to both.

breiler, 2023-02-27:
The settings made using the setup wizard are stored on the controller and not UGS. So the next time you connect to the machine it should be properly configured with step settings, homing, limit switches.

Some settings are however stored in UGS settings on your computer such as macros, probe settings, latest opened files, keymaps, window sizes, gamepad configuration and so on.
To make things complicated, there is a setting-file that is shared between "UGS Classic" and "UGS Platform" and there is a settings directory used exclusivly in "UGS Platform".

In order to run multiple instances of UGS you would need to run each instance with its own settings directory. This can be done by starting them with:
```
ugsplatform.exe --userdir %APPDATA\ugsplatform1ugsplatform.exe --userdir %APPDATA\ugsplatform1
```

breiler, 2023-02-27:
The settings that is made using the "Setup wizard" are stored on the controller and not UGS. So the next time you connect to the machine it should be properly configured with step settings, homing, limit switches.

Some settings are however stored in UGS settings on your computer such as macros, probe settings, latest opened files, keymaps, window sizes, gamepad configuration and so on.
To make things complicated, there is a setting-file that is shared between "UGS Classic" and "UGS Platform" and there is a settings directory used exclusivly in "UGS Platform".

In order to run multiple instances of UGS you would need to run each instance with its own settings directory. This can be done by starting them with:
```
ugsplatform.exe --userdir %APPDATA\ugsplatform1
ugsplatform.exe --userdir %APPDATA\ugsplatform2
```

But you would get into trouble with the shared settings which would be overwritten by each instance. This would probably eventually corrupt the settings...

So it could probably be done but I would recommend against it. And I don't have any plans on fixing this anytime soon.

chrisla23, 2023-02-27:
Running in a Docker container could be workable. I have done other serial port apps in a container on Windows before. 

-Chris

breiler, 2024-01-02:
Closing as we will not support this.

closed 2024-01-02 by breiler