winder/Universal-G-Code-Sender #2170: UGS GRBL1.1h with CNC Shield V4  Nano,no Potis reaktions for Override Feed and Speed at Analog in

hrenz, 2023-03-07:
### Version

2.0.14

### Hardware / Firmware

GRBL 1.1

### What happened

UGS latest version 
with GRBL1.1h at CNC Shield V4  Nano works fine,
but no Potis reaktions for Override Feed and Speed at the Analog inputs

### Operating System

win 10 UsB at Com 5   works fine

hrenz, 2023-03-07:
Hy,
cnc shield V4 with Arduino Nano
with GRBLV1.1h 

UGS  works , Steppers run

but at Analog Input A6, A7 are 2 Potis possible for Speed override and Feed Override
this doesn´t work 
no reaktion at UGS

breiler, 2023-03-07:
To my knowledge GRBL has no support for analog inputs, where did you read that?
What is a Potis - a potentiometer?

hrenz, 2023-03-08:
Yes 2 Potentiometer for Input  Speed and Feed 
at GRBL V1.1h at Analog A6, A7 with the CNC Shield V4

there are youtube vidoes and Pictures  and shematics

breiler, 2023-03-08:
Do you mind sharing the links to the videos, pictures and schematics?

hrenz, 2023-03-08:
Yes will do

[image]

breiler, 2023-03-08:
What is the source for these images?

I remain skeptical as there is nothing in the "standard" [GRBL](https://github.com/gnea/grbl/) firmware (that I can find) that refers to these pins or analog inputs. I never heard of such a feature...

This product wiki page doesn't mention the use of those pins:
https://wiki.keyestudio.com/Ks0152_keyestudio_A4988_3D_printer_stepper_motor_driver_CNC_shield_V4

What kind of potentiometer did you connect to those ports and how? 
Where did it say which values of potentiometer you should use? 
Are you really using a standard GRBL or a fork?

hrenz, 2023-03-08:
Its all for Ardino Nano with a 328

2 Potis with 10k

At cnc shield V4 clone are some bugs because there are error at DIR and STEP pins
so you must make a small fork

EstlCam use this 2 Poti too with the option GRBL V1.1 with PWM

[image]

[image]

closed 2023-03-08 by breiler

hrenz, 2023-03-08:
UGS works too with the Original CNC Shield V4 and Original GRBL V1.1.h,
but not with the 2 Potis for override Feed and Speed

CNC Shield V4 clone has a lot of Bugs, UGS cant work with it

hrenz, 2023-03-09:
OK, this is not so nice,  
Estlcam has many options for choosing different hardware. you can customize in detail, 
Set inputs, outputs, analog pins and the functions for them.
I was hoping that UGS can do that too

[image]

hrenz, 2023-03-09:
Have worked yesterday evening with UGS
GRBL V1.1h and Nano 328 with 4 Stepper

What I miss very much are ready made buttons 
with fixed functions, macros like homing, tool change, 
tool probe, workpiece measure. 
Cooling On, Suction On etc.

The normal user does not want to enter complicated G- and M-codes, he wants ready-made functions on a button, which can be set in a menu before.

What I also notice when the router is running,
The active G-code line in the program is not recognizable, it is not displayed

breiler, 2023-03-09:
Even if UGS would be able to configure and flash the Arduino with GRBL it would not be able to configure it to read values from potentiometers - beacuse the GRBL firmware does not support it! Please prove me wrong by pointing to the source code that does that in the firmware: https://github.com/gnea/grbl/

And even if GRBL did support it, it should have been implemented so that it changes the feed/speed internally in the controller transparent to the sender.

My german is a bit rusty, but in your latest screenshot it describes GRBL with PWM, there is nothing that says it is able to read from analog inputs, only PWM outputs.

You have still not provided sources to the first images describing the analog input ports. You said that you had links to the videos, pictures and schematics but have failed to provide them. Where did you get those images? Who is claiming that it should work?

breiler, 2023-03-09:
> What I miss very much are ready made buttons
with fixed functions, macros like homing, tool change,
tool probe, workpiece measure.
Cooling On, Suction On etc.
The normal user does not want to enter complicated G- and M-codes, he wants ready-made functions on a button, which can be set in a menu before.

I agree with you, but please create a new feature request or start a discussion as it is out of topic.

> What I also notice when the router is running,
The active G-code line in the program is not recognizable, it is not displayed

I don't understand what you mean.

hrenz, 2023-03-09:
at youtube CNC shield V4  with Poti override feed and speed
this one and the second one
https://www.youtube.com/watch?v=zGc6NAAxolI
https://www.youtube.com/watch?v=kjDy16rsqas

breiler, 2023-03-09:
He is not even using the GRBL firmware, he is using a GRBL shield and his own custom Arduino program. The links to his dropbox is broken which probably contains his software. Forget about using this in UGS...

Regarding highlighting the current sent gcode, are you using the toggle button described here: https://github.com/winder/Universal-G-Code-Sender/pull/2123

hrenz, 2023-03-09:
highlighting linie missiing at UGS

Example
[image]

hrenz, 2023-03-10:
sorry  think I have a Problem with chrome now
App CadasCAM is broken now

breiler, 2023-03-10:
And how is this related to UGS?