winder/Universal-G-Code-Sender #2174: ERROR [jSSC-2.8_x86_64.dll+0xb5db] con Windows 10 Home -- UGS no se abre

indipeers, 2023-03-10:
### Version

Other

### Hardware / Firmware

GRBL 0.8

### What happened

Hola buenas. Soy nuevo en el mundo del UGS y del CNC. He visto que varias personas han tenido también el problema del error [jSSC-2.8_x86_64.dll+0xb5db].
No entiendo muy bien porque me sale este error...les cuento un poco.
Mi antiguo ordenador murió (pobrecito mío, con lo bien que funcionaba) y compre un ordenador nuevo con el sistema operativo Windows 10 Home. Instale como siempre Arduino con su Grbl (Arduino y UGS es lo que me enlaza el CNC con el ordenador), instale la versión de Java Para UGS y finalmente instale UGS. Lo raro es que cuando instale todo esto en el ordenador nuevo con Win10, intente hacer una prueba y todo funciono genial.Pero al cabo de unos días, volví para cortar unas maderas y cuando abrí UGS y le di ha Open(que es cundo se buelca el Grbl en UGS) automaticamente el programa de UGS se cierra y me sale en el escritorio un archivo de error. Es el siguiente archivo:

[hs_err_pid12084.log](https://github.com/winder/Universal-G-Code-Sender/files/10943997/hs_err_pid12084.log)

No entiendo que la primera vez funcionara y luego sin hacer ningún cambio de nada, saliera este error. No se si Windows10 Home actualizó alguna cosa(como Java) y que por ese motivo ahora salga ese error...
Lo he intentado todo(dentro de mis conocimientos) : he formateado varias veces el ordenador y he reinstalado Windows, he probado varias versiones de UGS y de Java, he mirado de enable los Minidumps para Windows, pero el error no se soluciona... Estoy un poco desesperado ya que llevo mas de 2meses intentando buscar una solución a esto pero ya no se que mas puedo hacer y estoy algunos proyectos perdiendo de trabajo importantes....
La version de UGS que he utilizado siempre es un poco antigua creo , pero me iba genial. es la 2.0 nightly. Y la version de Grbl es la 0.8 . 
SI hay alguien que me pueda echar una mano para intentar solucionar este problema se lo agradecería muchísimo. :)

(debajo les dejo lo que muestra el archivo de error)

### How to reproduce

Archivo de error:

[hs_err_pid12084.log](https://github.com/winder/Universal-G-Code-Sender/files/10944074/hs_err_pid12084.log)

### Operating System

Windows 10 Home 64bits/ Portatil HP 840 G1 / version 22h2

breiler, 2023-03-12:
Difficult to tell, could be an antivirus program blocking in some way...

Does UGS Platform work? It is packaged with its Java, download the 2.0.14 version here:
https://github.com/winder/Universal-G-Code-Sender

indipeers, 2023-03-23:
@breiler perdona que no he dicho nada mas....estos días no tuve tiempo para hacer mas pruebas con el CNC. :_(
Solo decirte que instale la versión que me mandaste pero me surgieron nuevos problemas , ademas que la versión de UGS tenia algunos parámetros (como poner el proyecto en PAUSA) que no funcionaba bien y hacia que todo se bloqueara.
En todo caso mirare de abrir otra vez el problema en unos días. Así intentare mandar todas las dudas y preguntas que tenga de todo esto.
Aun así, muchísimas gracias por la ayuda @breiler  . Les digo algo en unos días. :)

breiler, 2023-03-23:
Just reopen if you have any new information

indipeers, 2023-03-28:
Hola @breiler ! te escribo por aquí solo para decirte que ya conseguí solucionar el problema. Al final utilicé la ultima versión de GRBL 1.1 y creo que la ultima versión de UGS Pataform, y ya todo funciona a la perfección. Era solo para darte las gracias por el soporte y la ayuda. Gracias por estar ahí! Saludos! :)

breiler, 2023-03-28:
Excellent, glad that it worked out for you!