winder/Universal-G-Code-Sender #2176: UGS won't load

ebetts62, 2023-03-14:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

program worked great... at first.  running on Liinux mint.  UGs version 2.0,0 on a Dell inspiron 10 laptop with intel core I3. now it doesnt completely load.
[image]
Tried reinstalling several times, but after one or two uses, it effectively freezes up at the point shown.  not sure what's causing it, but it seems to be after trying to use the visualizer.

### How to reproduce

Tried reinstalling several times, but after one or two uses, it effectively freezes up at the point shown. not sure what's causing it, but it seems to be after trying to use the visualizer.

### Operating System

linux mint

breiler, 2023-03-15:
Please try UGS 2.0.14 or later.

ebetts62, 2023-03-16:
> Please try UGS 2.0.14 or later.

found 2.0.13... didn't see .14.  seems to be working, so far...  thanks!

closed 2023-03-18 by ebetts62