winder/Universal-G-Code-Sender #2195: Hangs on start

risototh, 2023-04-08:
### Version

Other

### Hardware / Firmware

Other (add details about it below)

### What happened

I tried to compile fresh pull from master (last commit 94b3158a6d0857db192e91cea7a4d78296721793, Fri Apr 7 14:08:06 2023 +0200) to be able to contribute some enhancements in the future, but I have a problem with starting the application (platform). Finaly the app compiles and starts with the splash screen, then the main window appears, but without the tabs inside. Just a gray window, with the toolbar and without the menu. Then beachball on Mac... The log file is attached. Any ideas?

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/11183445/messages.log)
[image]

### How to reproduce

Just compiled and ran from Netbeans, or from CLI by the command provided in `README.md`, or by the `run_platform.sh` - everytime the same behaviour.

### Operating System

macOS 10.15.7

### Anything else

Compiled in Netbeans 17 with the OpenJDK 13.0.1+9

risototh, 2023-04-08:
OK, I tried the generated ZIP app package on another Mac, and it just works... Copied the settings from ~/Library/Preferences/ugs back to my dev computer - same issue. At the end and few hours, I tried the restart, and now it works... But why? What changed? I'm attaching the new `messages.log` after the successfull start.
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/11184000/messages.log)

breiler, 2023-04-09:
Looking through the logs, there are no errors in them so there is not much to work on. 

I know that there was a similar issue on Linux where a popup was shown beneath the startup splash and also blocking it from every finishing. So I loaded the popup in its own thread:
https://github.com/winder/Universal-G-Code-Sender/commit/22199138c0ae7914c57146313f1534e59806afa5#diff-f4eb73bbebf967f9ffb977b16ac21ec0bc55fb27c6981312713e498d9844347aR102

But other than that I have no clue what is causing this.