winder/Universal-G-Code-Sender #2214: Windows disapeared

Yonaskassa, 2023-05-05:
### Version

2.0.17

### Hardware / Firmware

GRBL 1.1

### What happened

Dears,
Why windows disapeared as per attached image and how to repair these issue.

[image]

breiler, 2023-05-05:
Can you attach your log file located here:
C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log

Yonaskassa, 2023-05-05:
Here is attached.
I have two ugs versions on my laptop, Is it have a problem?
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/11409072/messages.log)

breiler, 2023-05-06:
It should be ok to have multiple versions, but only one can be open at a time.

I see a couple of errors in the log but those should not make the windows disappear. The settings may have been corrupt, try and delete all files under these directories:
```
C:\Users\Yonas.kassa\AppData\Local\ugsplatform\Cache\var\cache
C:\Users\Yonas.kassa\AppData\Roaming\ugsplatform
```

Yonaskassa, 2023-05-06:
Dear,

Thanks now displayed.

Regards!

closed 2023-05-06 by Yonaskassa