winder/Universal-G-Code-Sender #2221: UGS do not continue when connecting to Arduino mega with Grbl

cgerrist, 2023-05-13:
### Version

2.0.17

### Hardware / Firmware

GRBL 1.1

### What happened

After installing the latest version, UGS do not respond when I make a connection to a Arduino Mega with lastes Grbl software.
I need to do a hardware reset then it makes connection. First I thought it has something to do with de Grbl software. I tested also with 'Better GRBL' via octoprint. This software has no issues.
What I see is the the RX light flashes every second.
After a hardware reset it works but sometimes the Arduino gives an boot error.
When it works the rx and tx light flickers every half a second.

### How to reproduce

Connect to a Arduino Mage with the latest Grbl and make a connection.
When It hang give an hardware reset. Them it continue. 

I use a Mac mini M2 but I had this problem also with an x86 Mac

### Operating System

I have a Mac mini M2 with the latest installation. 

### Anything else

It happens every time when I connect.

GettingRusty, 2023-05-14:
I have the same issue, tried multiple Mega boards, and Windows 10 machines and Macosx. Always needs the Mega reset pressed to connect. Connects to UNO’s with no need to press reset.

breiler, 2023-05-14:
Exactly which version of GRBL are you running on the Mega?
https://github.com/gnea/grbl-Mega
https://github.com/fra589/grbl-Mega-5X

> After a hardware reset it works but sometimes the Arduino gives an boot error.

What does the error say?

@GettingRusty are you also running version 2.0.17? Earlier versions of UGS could not connect properly with the megas but should work as of 2.0.17.

GettingRusty, 2023-05-14:
Thanks, I’m trying the grbl-mega. I did not know about grbl-mega-5x. I should check it out.
I’ve been using version 2.0.14 of UGS.
I see there is now a version 2.0.15. Where do I find 2.0.17 to try?

breiler, 2023-05-15:
@GettingRusty https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.17

GettingRusty, 2023-05-15:
Thanks,no go. Did version 2.0.17 on 2 windows 10 machines, a Macbook pro M1 with 4 different Mega boards.
I used these:
https://github.com/gnea/grbl-Mega
https://github.com/fra589/grbl-Mega-5X

All do an initial connect and they respond to three commands but then Comms drops and the system locks up with not able to use the serial port again. A UGS restart and I can try again. 

Gone back to version 2.0.15 at least on using the board reset button the Megas respond and UGS stays working.

breiler, 2024-11-12:
A lot of work has been done to the new initiation protocol for GRBL and I can't reproduce this problem.

closed 2024-11-12 by breiler