winder/Universal-G-Code-Sender #2222: COM port connected but not active

Yonaskassa, 2023-05-14:
### Version

2.0.15

### Hardware / Firmware

GRBL 1.1

### What happened

Dear @breiler,

Why ugs connected but not active, I choose the right port and baud rate. I tried by Arduino IDE and its working properly I can Uploade the code without a communication problem, But When I use UGS some times connected and active. But most of the time it will not active. I am tired to get the exact problem Please support me. Is there any windows problem or setting to deactive the com port when I use UGS?

breiler, 2023-05-14:
What hardware are you using, Arduino UNO or Arduino Mega?

Can you try and connect to the controller using the Arduino IDE:s serial monitor. After you connect you should get a welcome string. What does that say?

Yonaskassa, 2023-05-14:
I am using Arduino Mega 2560,
Here is attached well come string after connecting arduino IDE.
But when am connecting to UGS its connected but not active.
[image]

Yonaskassa, 2023-05-14:
And here is attached what UGS saying.
Thank you for your usual support!

[image]
[image]

breiler, 2023-05-15:
This is not what I meant, I wanted you to use the Arduino IDE Serial Monitor to get the output from GRBL.

You now need to reflash GRBL to your controller, and since you are using an Arduino Mega you will have to make sure that you are using the GRBL suited for those controller board:
https://github.com/gnea/grbl-Mega or https://github.com/fra589/grbl-Mega-5X/

breiler, 2023-05-15:
And you will need to use UGS 2.0.17 or later which contains a fix that is required to connect to Mega Controllers: https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.17

The nightly build also contains a fix for this that you requested previously:
https://github.com/winder/Universal-G-Code-Sender/issues/2209

The nightly version can be downloaded from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

Yonaskassa, 2023-06-16:
@breiler 
When I connect to Ugs the error displayed on Console which says '' An error was detected while sending '?': +CME ERROR: 100. Streaming has been paused. ''
And it disconnected by it self.

breiler, 2023-06-16:
`+CME ERROR: 100` is nothing that comes from UGS or GRBL. Are you sure that you've connected to  your GRBL device and not a 3G/4G internet dongle?

Yonaskassa, 2023-06-17:
@breiler 
Dear Sorry for my mistake, after many trial I got the problem by relating with other laptop working ugs.
I found the zip file didnot work downloaded on the recent 2023, and I took from other pc downloaded Ugs zip file on 2022 and its working smoothly, and know I do have 2 versions smoothly working v 2.0.14 and snapshot 2.0 03/09/2022. Thank You so much🙏

closed 2023-07-03 by breiler