winder/Universal-G-Code-Sender #2225: UGS not running on Raspberry Pi 4 with 4GB and Ubuntu 22.04.02

FionNoir, 2023-05-27:
### Version

2.0.14

### Hardware / Firmware

GRBL 1.1

### What happened

Application does not start cause of following error:
`ugsplatform-pi/jdk/jdk8u275-b01-jre/bin/java: no such file or folder`

But the file java is in the folder.
I've tried JDK 8 and JDK 11 without success. I don't understand the problem here.

### How to reproduce

Just try to launch UGS on a Pi with Ubuntu

### Operating System

Ubuntu for Pi

breiler, 2023-05-29:
Are you using a 32-bit version of the OS?
https://github.com/winder/Universal-G-Code-Sender/wiki/Installing

mlwash, 2023-06-11:
I am having the same problem.  I have tried JDK 8 and JDK 11 without success.  I have also posted an issue on Open JDK and a suggestion came to try Open JDK 1.8.0_362, but there are not Raspberry version of this.

breiler, 2023-10-15:
The nightly build now have a Linux ARM64 version, please give it a try and report back
https://github.com/winder/Universal-G-Code-Sender/releases/tag/nightly

closed 2024-01-23 by breiler