winder/Universal-G-Code-Sender #2230: UGS stopped working on Raspberry Pi4 after last OS update

mlwash, 2023-06-07:
### Version

2.0.17

### Hardware / Firmware

GRBL 1.1

### What happened

I have been using UGS for about 2 years without any problems.  I have a Raspberry Pi4 and an Auduino Nano for the GRBL. 

After the last Raspberry update, I continue to get the following error while executing UGS:

# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGBUS (0x7) at pc=0xf6196f20, pid=1246, tid=0xf5fb8440
#
# JRE version:  (8.0_275-b01) (build )
# Java VM: OpenJDK Client VM (25.275-b01 mixed mode linux-aarch32 )
# Problematic frame:
# V  [libjvm.so+0x16ef20]  ClassFileParser::parse_constant_pool_entries(int, Thread*)+0x47c
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again

I have tried updating my Raspberry OS with a new SD card, but get the same result.  
I have also downloaded the latest version of UGS and continue to get the same result.

Attached is the error report,
[hs_err_pid1246.log](https://github.com/winder/Universal-G-Code-Sender/files/11680945/hs_err_pid1246.log)

### Operating System

Raspberry Pi OS Full (32 bit)

### Anything else

It is happening all the time now.

skobls, 2023-07-01:
I have an identical problem on the same hardware. However, I was able to find a workaround. I installed the JDK: `sudo apt install default-jdk`, and used the "All platforms (requires Java)" release option instead of "Raspberry Pi"

AbitEncrypted, 2023-09-06:
I also have an identical problem. Following @skobls instructions I can get UGS to run.  However, the only port that is presented is 'ttyAMA0' which UGS attempts a connection on but fails.  There is no 'ttyUSB0' available.
I have tested the workaround using a bare metal fresh install of the recommended 32 bit Raspberry Pi OS (bullseye) on a Pi 4B
Any suggestions will be well received.

AbitEncrypted, 2023-09-07:
Whereas @skobls has a good workaround I thought I would attempt to get a more sustainable solution. Partly because the 'All Platforms' solution resulted in a number of warnings before UGS ran.
I decided to use Raspberry Pi OS 64bit (bullseye). Most googling suggested that UGS is better suited to a 32bit platform but what the hell.
So having installed the 64bit bullseye I set about trying various builds of UGS, Linux 2.0.18/20; RaspberryPi 2.0.18/20 none of which worked and gave the same errors as on the 32bit RPi.
I then cloned the github repository on the Pi

`git clone https://github.com/winder/Universal-G-Code-Sender`

Installed maven
```
apt-get update
apt-get install maven

```
Built an image from the Universal-G-Code-Sender folder created by the cloning.

`./run_platform.sh`

The build went smoothly with no reported errors.

Great news, I have a working UGS. 
This gave me the confidence to move the OS and UGS to a USB SSD which now gives me a very responsive and fast solution.

breiler, 2023-09-07:
The UGS-packages for RaspberryPI currently only comes with a ARM 32-bit Java environment. So that is the reason why those packages won't run on a ARM 64-bit OS. I have no idea why it has stopped working on the 32-bit environment as well.

When I experimented with a 64-bit packaging I never got usable performance on my old RPI 3b, so that is why those packages doesn't exist. 

You can, with a bit of fiddeling, change the `conf/ugsplatform.conf` to point to your own Java instead of using the "run_platform.sh"-script. That script will recompile the software everytime.

I now have a RPI 4b on my desk waiting for me to get working on this again...

stephenpapierski, 2023-09-17:
I'll also note that I've had issues with the RaspberryPi version on both 32-bit and 64-bit OS versions. Combining some of the above steps, I have a working binary on the 64-bit version of the OS on a Raspberry Pi 4.

1. Install Requirements
```
sudo apt update
sudo apt install default-jdk maven
```
2. Clone and Build
```
git clone https://github.com/winder/Universal-G-Code-Sender
cd Universal-G-Code-Sender
./run_platform.sh
```
3. Run Application
```
ugs-platform/applications/target/ugsplatform/bin/ugsplatform
```

Ka8oum, 2023-09-29:
Hi,

Found an easy workaround for this issue.
edit the /boot/config.txt
in the [pi4] section add "arm_64bit=0"

this should look something like this:

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2
arm_boost=1
arm_64bit=0

then reboot and start UGS

this is because in one of the latest updates, raspbian uses the 64bit linux kernel, even on the 32bit OS

Hope this helps

Borris5163, 2023-10-08:
Thank you @Ka8oum , you solution worked for me, much appreciated.

breiler, 2023-10-13:
The nightly build now have a Linux ARM64 version, please give it a try and report back
https://github.com/winder/Universal-G-Code-Sender/releases/tag/nightly

breiler, 2024-01-02:
Closing as we didn't get a response from the user.