winder/Universal-G-Code-Sender #2258: Incompatible with dxf2gcode exports

timint, 2023-07-17:
### Version

2.0.18

### Hardware / Firmware

GRBL 1.1

### What happened

Loaded a gcode file from dxf2gcode. UGS says coordinates are invalid because of whitespace after the axis letter.

### How to reproduce

Produce a GCode file containing whitespace in coordinates after the axis letter. E.g. "X   10.00"

### Operating System

Windows 11

breiler, 2023-07-17:
I just tried it with a sample file found online somewhere and the only place that I get errors are in the editor. Was this what you meant?

[image]

timint, 2023-07-17:
Yes. Thank you for providing a screen shot. I was on mobile.

breiler, 2023-07-17:
When using GRBL it will strip white spaces by default before sending it, so your regexp example should work just fine for viewing the gcode.

I'll make the adjustments to the parser, but that will take a couple of days since I am out of town.

Thanks for the bug report

timint, 2023-07-17:
No rush, I'm just happy to bring bones to the table. :)

closed 2023-08-21 by breiler

breiler, 2023-08-21:
It should now be available in the latest nightly build here:
https://github.com/winder/Universal-G-Code-Sender#downloads

PR #2294 (2023-08-21): Gcode parser now allows multiple spaces and tabs
Gcode parser can not parse multiple white spaces and tabs between parameter and value.

Fixes #2258