winder/Universal-G-Code-Sender #2265: The autoleveler plugin stopped working in 2.0.18

pherrera2, 2023-07-26:
### Version

2.0.18

### Hardware / Firmware

GRBL 1.1

### What happened

It does the surface scanning but nothing is shown in the visualizer, if I click "View Data" the array is empty or, at most, has a single point., 
The process works properly, ie. the probe is definitelly working in the surface, moving, etc. but no points gets recorded.

### How to reproduce

I just define an area in the autoleveler and run it, the array will be empty or just have one or two points, worked fine in 2.0.17

### Operating System

Windows 10 22H2

pherrera2, 2023-07-26:
The probe module works fine, I can probe individual points, this is limited to the autoleveler

breiler, 2023-08-18:
Please try with version 2.0.19, make sure to double check your settings after upgrading:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.19

pherrera2, 2023-08-22:
Works perfectly on 2.0.19! thanks! sorry for delay it's a DIY machine and I had to assemble some stuff

closed 2023-08-22 by breiler

gugungunawandotcom, 2023-09-04:
apakah fittur autoleveler tidak dapat bekerja dengan baik di ugs versi2.0.7 SNAPSHOT/Dec 02.2020?tapi sewaktu dial di titik titik resolusi nya bisa dibaca, sedangkan saat di run kan hanya putar di satu titikdengan tidak menyentuh media dan tidak berpindah ke titik yang lainnya.

gugungunawandotcom, 2023-09-04:
is the autoleveler feature not working properly in ugs version 2.0.7 SNAPSHOT/Dec 02.2020? but if the dial is on the dots then the resolution can be read, whereas if it is run it can only be rotated at one point without touching the media and not moving to another point

breiler, 2023-09-04:
@gugungunawandotcom I have no idea what you are saying. I have made a lot of fixes to the autoleveler plugin which is only available in version 2.0.19 or higher. Please upgrade and try again. If it still doesn't work post a new issue.

gugungunawandotcom, 2023-09-05:
ok.tq.i'll try to update first