winder/Universal-G-Code-Sender #2278: Probe Module window size persistent

itsnotlevel1, 2023-08-13:
### Version

2.0.18

### Hardware / Firmware

FluidNC

### What happened

I didnt notice until I started customizing the UGS screen, that the Probe module window is persistent in its size. It wont let me adjust.....autoleveler and dowel maker all of those plug ins resize manually and automatically when I move things around. Probe, no dice. 

### How to reproduce

I just open up the probe module and its the same size any where I put it. And it wont resize. 

### Operating System

Linux Mint 20.2

breiler, 2023-08-16:
The probe UI should now be able to shrink as of #2280. It is now available in the latest nightly build. 

Please be careful if you try this out. I feel that I have done a lot of testing, but you can never test enough...

itsnotlevel1, 2023-08-18:
Nightly build downloaded today and now probe module doesnt open at all....window doesnt show...all other modules open. Reset windows....still nada. :-(

breiler, 2023-08-18:
Thanks for the feedback. Could you post the the logs. They are located here: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

itsnotlevel1, 2023-08-18:
Logs Attached... :)
[UGSMessagesLog8182023.txt](https://github.com/winder/Universal-G-Code-Sender/files/12383734/UGSMessagesLog8182023.txt)

breiler, 2023-08-18:
Yep, I found the problem, thanks!
I'll get a fix out tomorrow.

itsnotlevel1, 2023-08-18:
You rock! I contributed today. I hope you guys keep up the good work. :thumbsup:

Paul

breiler, 2023-08-19:
This was caused by the settings panel for probing where it tried to set a value to one of the fields with a lower value than allowed. The minimum value was set to `1`. 

1 inch per minute might be slow enough, but I changed this so the lowest value is `0.1`.

If the settings file contains lower values it will automatically set the lowest allowed value instead of crashing the module.

The changes is now available in the latest nightly build:
https://github.com/winder/Universal-G-Code-Sender#downloads

The other problem that you reported #2287 might be related to this bug.

itsnotlevel1, 2023-08-19:
I downloaded the latest nightly and yes everything resizes now, thank you!! It doesnt seem to be related to the #2287 bug though....in other words it didnt change anything from #2287 ....ill test it some more.

breiler, 2023-08-21:
I am closing this as the panels should resize a bit better now. Just reopen if you still have problems.

closed 2023-08-21 by breiler