winder/Universal-G-Code-Sender #2287: Over shooting x-axis for probing

itsnotlevel1, 2023-08-18:
### Version

2.0.19

### Hardware / Firmware

FluidNC

### What happened

I have been trying to make the probe function work and it actually only worked ONCE...lol. Now whenever using the xyz probe function: Z works great....X works great.....but when the spindle moves over the touch the Y side, it over shoots the probe by at least 1 inch.....??? Video below. 

https://drive.google.com/file/d/1dL07ceMukKIUcAebgRDHZZhMMcU-098Y/view?usp=sharing

### How to reproduce

Just run probe again. 

### Operating System

Linux Mint 20.2

### Anything else

everytime

breiler, 2023-08-19:
Can you screenshot your your probe settings.

itsnotlevel1, 2023-08-19:
[image]
[image]

breiler, 2023-08-19:
Thanks, I see the problem now when trying the same settings on my machine. It was hardly noticeable when using "smaller" values so I never caught this. It is getting late here, so I'll take a look at it tomorrow morning.

itsnotlevel1, 2023-08-19:
Yes sir .... Please take your time, it's the weekend. 😁. Btw.... Those are imperial measurements just incase I didn't specify. Thanks!

breiler, 2023-08-20:
I think that I figured it out. It is available in the latest nightly build:
https://github.com/winder/Universal-G-Code-Sender#downloads

itsnotlevel1, 2023-08-21:
Yayyyy!! You are a coding god! Thank you for your time.

breiler, 2023-08-21:
Excellent, thanks for the feedback. Is it ok to close the issue?

itsnotlevel1, 2023-08-21:
Yes, I believe so.

Paul

-------- Original Message --------