winder/Universal-G-Code-Sender #2293: Errors sen in the editor

fbonin5491, 2023-08-21:
### Version

2.0.19

### Hardware / Firmware

GRBL 1.1

### What happened

UGS can't send the code because it find errors everywhere.
here is the code :
(Generated with: DXF2GCODE, Version: Py3.6.7 PyQt5.11.3, Date: $Date: Thu Jan 3 16:29:29 2019 +0100 $)
(Created from file: Z:/home/bobphils/Bobphils2.dxf)
(Output format description: G-CODE for LinuxCNC)
(Time: Sat Aug 19 17:52:29 2023)
G21 (Units in millimeters)
G90 (Absolute programming)
M3
G0 Z  15.000 

(*** LAYER: noname ***)

S6000

(* SHAPE Nr: 1 *)
G0 X   3.750 Y  10.806
M3 M8
G0 Z   3.000 
F250
G1 Z  -1.000
G1 X   6.528 Y  10.806
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 2 *)
G0 X   3.750 Y   5.250
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X   3.750 Y  15.250
G1 X   6.528 Y  15.250
G2 X   6.528 Y  10.806 I   0.000 J  -2.222
G2 X   6.528 Y   5.250 I   0.000 J  -2.778
G1 X   3.750 Y   5.250
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 3 *)
G0 X  12.639 Y   7.472
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X  12.639 Y   9.694
G2 X  17.083 Y   9.694 I   2.222 J   0.000
G1 X  17.083 Y   7.472
G2 X  12.639 Y   7.472 I  -2.222 J   0.000
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 4 *)
G0 X  20.417 Y  11.917
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X  23.194 Y  11.917
G2 X  24.861 Y  10.250 I   0.000 J  -1.667
G1 X  24.861 Y   6.917
G2 X  23.194 Y   5.250 I  -1.667 J  -0.000
G1 X  20.417 Y   5.250
G1 X  20.417 Y  15.250
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 5 *)
G0 X  28.194 Y   1.917
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X  28.194 Y  11.917
G1 X  30.972 Y  11.917
G2 X  32.639 Y  10.250 I  -0.000 J  -1.667
G1 X  32.639 Y   6.917
G2 X  30.972 Y   5.250 I  -1.667 J   0.000
G1 X  28.194 Y   5.250
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 6 *)
G0 X  35.972 Y  15.250
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X  35.972 Y   5.250
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 7 *)
G0 X  35.972 Y  11.917
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X  38.750 Y  11.917
G2 X  40.417 Y  10.250 I   0.000 J  -1.667
G1 X  40.417 Y   5.250
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 8 *)
G0 X  43.750 Y  11.917
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X  43.750 Y   5.250
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 9 *)
G0 X  43.750 Y  15.250
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G1 X  43.750 Y  14.694
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 10 *)
G0 X  48.194 Y   5.250
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G2 X  47.083 Y   6.361 I  -0.000 J   1.111
G1 X  47.083 Y  15.250
G1 Z   3.000
G0 Z   5.000 
M9 M5

(* SHAPE Nr: 11 *)
G0 X  55.417 Y  11.361
M3 M8
G0 Z   3.000 
G1 Z  -1.000
G3 X  52.917 Y  11.917 I  -2.515 J  -5.415
G3 X  52.313 Y   9.277 I  -0.005 J  -1.388
G1 X  55.187 Y   7.890
G2 X  54.583 Y   5.250 I  -0.604 J  -1.251
G2 X  51.528 Y   5.806 I  -0.002 J   8.668
G1 Z   3.000
G0 Z   5.000 
M9 M5
G0 X   0.000 Y   0.000
M5
M2 (Program end)

### Operating System

Linux Mint Vera , Mate

### Anything else

Works perfectly with bCNC , except that i have to use GRBL 0.9

breiler, 2023-08-21:
Duplicate of #2258

closed 2023-08-21 by breiler

breiler, 2023-08-21:
It should now be available in the latest nightly build here:
https://github.com/winder/Universal-G-Code-Sender#downloads

fbonin5491, 2023-08-22:
This built seems to work fine , thanks a lot