winder/Universal-G-Code-Sender #2295: ugs drops serial connection to grbl1.1 after saying it connected

wyzarddoc, 2023-08-21:
### Version

2.0.19

### Hardware / Firmware

GRBL 1.1

### What happened

UGS starts shows connection /dev/ACM0 I click on the connect icon and it flashes says it connected but no red reset and then connection icon shows open connection
Using JDK-17

### How to reproduce

download UGS unzip to directory installed jdk17-sdk and jdk17-sre 
go to the unzipped directory plug in Pico with preloaded grbl1.1 
start UGS then try to connect. One time got option to connect to /dev/ttyS4

### Operating System

ubuntu 22.04 Universal Gcode Platform 20230126

### Anything else

/home/doc/.cache/ugsplatform/var/cache/all-clusters.dat
/home/doc/.cache/ugsplatform/var/cache/all-files.dat
/home/doc/.cache/ugsplatform/var/cache/all-installer.dat
/home/doc/.cache/ugsplatform/var/cache/all-layers.dat
/home/doc/.cache/ugsplatform/var/cache/all-local-layers.dat
/home/doc/.cache/ugsplatform/var/cache/all-manifests.dat
/home/doc/.cache/ugsplatform/var/cache/all-modules.dat
/home/doc/.cache/ugsplatform/var/cache/all-resources.dat
/home/doc/.cache/ugsplatform/var/cache/localeVariants
/home/doc/.cache/ugsplatform/var/cache/package-attrs.dat
/home/doc/.cache/ugsplatform/var/cache/lastModified/all-checksum.txt

wyzarddoc, 2023-08-21:
This is on lenovo and HP laptops on my HP desktop seems to connecdt but get $X error

wyzarddoc, 2023-08-21:
will do in little while thanks

wyzarddoc, 2023-08-21:
breiler this is on a ubuntu linux machine and I can't find a messages.log

wyzarddoc, 2023-08-21:
> Can you post your log file from:
> ~/.ugsplatform/var/log/messages.log

I am using this on ubuntu linux and cannot find the log directory or message files thanks
Web-design, Woodworking, 3D-Art

breiler, 2023-08-22:
You wrote that you were using `Version 2.0.19` and then `Universal Gcode Platform 20230126`. So I really don't know which version you are running.

You wrote that you installed Java 17, this is not needed if you use the package specific for Linux where Java comes bundled. But if you choose the "All platforms" package you need to install Java separatley. But you would need Java 13, as the higher versions contains language changes which are not compatible with UGS yet.

So instead of me asking hundreds of questions, the log files would tell me all of that.

Do you have a `~/.ugsplatform` directory?

wyzarddoc, 2023-08-30:
will look for the file and send it along. Please note I am running this under Ubuntu 20.04 in case the files are different. Could be the linux distribution since I had to sudo apt-get autoremove brltty to get the serial to connect to UGS using the ACM0 port. Sorry for the delay in replying

wyzarddoc, 2023-08-30:
O.K. false alarm ugs seems to be working today!! after ubuntu os updates. Will try with machine that controls the cnc and get back to you if there are problems that persist thanks for the reply

wyzarddoc, 2023-08-31:
Stopped again her are the log files
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/12481842/messages.log)

breiler, 2023-08-31:
Morning, thanks for the log. From that it looks like you are running grblHAL and not GRBL 1.1 and UGS 2.0.18 and not 2.0.19.

This error explains one problem when if you unplug the controller and plug it in again, Linux will give it a new port (it changed from ttyACM0 to ttyACM1). I see that you found the new port was able to connect the second time:
```
INFO [com.willwinder.ugs.nbp.core.actions.ConnectDisconnectAction]: openclose button, connection open: false
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Connecting to GRBL on port ttyACM0
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Exception in openCommConnection.
java.io.IOException
	at com.fazecast.jSerialComm.SerialPort.getCommPort(SerialPort.java:407)
Caused: com.fazecast.jSerialComm.SerialPortInvalidPortException: Unable to create a serial port object from the invalid port descriptor: /dev/ttyACM0
	at com.fazecast.jSerialComm.SerialPort.getCommPort(SerialPort.java:410)
	at com.willwinder.universalgcodesender.connection.JSerialCommConnection.initSerialPort(JSerialCommConnection.java:67)
	at com.willwinder.universalgcodesender.connection.JSerialCommConnection.setUri(JSerialCommConnection.java:43)
Caused: com.willwinder.universalgcodesender.connection.ConnectionException: Couldn't parse connection string jserialcomm://ttyACM0:115200
	at com.willwinder.universalgcodesender.connection.JSerialCommConnection.setUri(JSerialCommConnection.java:45)
	at com.willwinder.universalgcodesender.connection.ConnectionFactory.getConnection(ConnectionFactory.java:47)
	at com.willwinder.universalgcodesender.communicator.AbstractCommunicator.connect(AbstractCommunicator.java:74)
	at com.willwinder.universalgcodesender.communicator.BufferedCommunicator.connect(BufferedCommunicator.java:289)
	at com.willwinder.universalgcodesender.GrblController.openCommPort(GrblController.java:149)
[catch] at com.willwinder.universalgcodesender.model.GUIBackend.openCommConnection(GUIBackend.java:759)
	at com.willwinder.universalgcodesender.model.GUIBackend.connect(GUIBackend.java:186)
	at com.willwinder.ugs.nbp.core.actions.ConnectDisconnectAction.lambda$connect$0(ConnectDisconnectAction.java:126)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
INFO [FirmwareAction]: Changed to firmware GRBL
INFO [UGSEventDispatcher]: Adding UGSEvent listener: MacroAction$$Lambda$248/0x0000000800d33440
INFO [com.willwinder.ugs.nbp.core.actions.ConnectDisconnectAction]: openclose button, connection open: false
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Connecting to GRBL on port ttyACM1
```

This problem is related to a bug where UGS is treating a message from grblHAL as if it were a status message from the GRBL 0.9 protocol. It should have been fixed as of version 2.0.19 of UGS.
```
SEVERE [com.willwinder.universalgcodesender.AbstractController]: Problem parsing command: GrblCommand[command=FIRMWARE:grblHAL,commandNum=-1,comment=,id=9,isDone=false,isError=false,isGenerated=true,isOk=false,isSent=false,isSkipped=false,isTemporaryParserModalChange=false,listeners=<null>,originalCommand=<null>,response=<null>]
com.willwinder.universalgcodesender.gcode.util.GcodeParserException: Multiple F-codes on one line.
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processCommand(GcodeParserUtils.java:114)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:143)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:130)
[catch] at com.willwinder.universalgcodesender.AbstractController.updateParserModalState(AbstractController.java:855)
	at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:238)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:769)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.lambda$rawResponseListener$0(CommunicatorEventDispatcher.java:60)
	at java.base/java.lang.Iterable.forEach(Unknown Source)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.rawResponseListener(CommunicatorEventDispatcher.java:60)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.dispatchEvent(AsyncCommunicatorEventDispatcher.java:126)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.run(AsyncCommunicatorEventDispatcher.java:78)
	at java.base/java.lang.Thread.run(Unknown Source)
WARNING [com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils]: An unknown gcode command was detected in: G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.
```

So I think that the solution is to upgrade to v2.0.20:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.20

wyzarddoc, 2023-08-31:
Thanks so much!!!  V2.0.20 works ok . Still get a alarm condition on start-up. Is there a way to write a configuration file to stop that?
Web-design, Woodworking, 3D-Art 

 
 

https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.20

wyzarddoc, 2023-08-31:
Thanks again!!!
*** Connecting to jserialcomm://ttyACM0:115200>>> ?>>> okok>>> $I[VER:1.1f.20230825:][OPT:VNMSL,35,1024,3,0][AXS:3:XYZ][NEWOPT:ENUMS,RT+,ES,SED,RTC][FIRMWARE:grblHAL][NVS STORAGE:*FLASH][FREE MEMORY:226K][DRIVER:RP2040][DRIVER VERSION:230820][DRIVER OPTIONS:SDK_1.5.0][BOARD:https://github.com/phil-barrett/PicoCNC][AUX IO:4,8,0,1]ok>>> $$$0 = 10.0    (Step pulse time, microseconds)$1 = 25    (Step idle delay, milliseconds)$2 = 0    (Step pulse invert, mask)$3 = 0    (Step direction invert, mask)$4 = 7    (Invert step enable pin, boolean)$5 = 0    (Invert limit pins, boolean)$6 = 0    (Invert probe pin, boolean)$9 = 1   $10 = 511    (Status report options, mask)$11 = 0.010    (Junction deviation, millimeters)$12 = 0.002    (Arc tolerance, millimeters)$13 = 0    (Report in inches, boolean)$14 = 0   $15 = 0   $16 = 0   $17 = 0   $18 = 0   $19 = 0   $20 = 0    (Soft limits enable, boolean)$21 = 0    (Hard limits enable, boolean)$22 = 0    (Homing cycle enable, boolean)$23 = 0    (Homing direction invert, mask)$24 = 25.0    (Homing locate feed rate, mm/min)$25 = 500.0    (Homing search seek rate, mm/min)$26 = 250    (Homing switch debounce delay, milliseconds)$27 = 1.000    (Homing switch pull-off distance, millimeters)$28 = 0.100   $29 = 0.0   $30 = 1000.000    (Maximum spindle speed, RPM)$31 = 0.000    (Minimum spindle speed, RPM)$32 = 0    (Laser-mode enable, boolean)$33 = 5000.0   $34 = 0.0   $35 = 0.0   $36 = 100.0   $37 = 0   $39 = 1   $40 = 0   $43 = 1   $44 = 4   $45 = 3   $46 = 0   $62 = 0   $63 = 3   $64 = 0   $65 = 0   $100 = 250.000    (X-axis travel resolution, step/mm)$101 = 250.000    (Y-axis travel resolution, step/mm)$102 = 250.000    (Z-axis travel resolution, step/mm)$110 = 500.000    (X-axis maximum rate, mm/min)$111 = 500.000    (Y-axis maximum rate, mm/min)$112 = 500.000    (Z-axis maximum rate, mm/min)$120 = 10.000    (X-axis acceleration, mm/sec^2)$121 = 10.000    (Y-axis acceleration, mm/sec^2)$122 = 10.000    (Z-axis acceleration, mm/sec^2)$130 = 200.000    (X-axis maximum travel, millimeters)$131 = 200.000    (Y-axis maximum travel, millimeters)$132 = 200.000    (Z-axis maximum travel, millimeters)$341 = 0   $342 = 30.0   $343 = 25.0   $344 = 200.0   $345 = 200.0   $346 = 1   $370 = 0   $372 = 0   $384 = 0   $398 = 35   $481 = 0   $484 = 1   ok>>> $G[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]ok>>> $$$0 = 10.0    (Step pulse time, microseconds)$1 = 25    (Step idle delay, milliseconds)$2 = 0    (Step pulse invert, mask)$3 = 0    (Step direction invert, mask)$4 = 7    (Invert step enable pin, boolean)$5 = 0    (Invert limit pins, boolean)$6 = 0    (Invert probe pin, boolean)$9 = 1   $10 = 511    (Status report options, mask)$11 = 0.010    (Junction deviation, millimeters)$12 = 0.002    (Arc tolerance, millimeters)$13 = 0    (Report in inches, boolean)$14 = 0   $15 = 0   $16 = 0   $17 = 0   $18 = 0   $19 = 0   $20 = 0    (Soft limits enable, boolean)$21 = 0    (Hard limits enable, boolean)$22 = 0    (Homing cycle enable, boolean)$23 = 0    (Homing direction invert, mask)$24 = 25.0    (Homing locate feed rate, mm/min)$25 = 500.0    (Homing search seek rate, mm/min)$26 = 250    (Homing switch debounce delay, milliseconds)$27 = 1.000    (Homing switch pull-off distance, millimeters)$28 = 0.100   $29 = 0.0   $30 = 1000.000    (Maximum spindle speed, RPM)$31 = 0.000    (Minimum spindle speed, RPM)$32 = 0    (Laser-mode enable, boolean)$33 = 5000.0   $34 = 0.0   $35 = 0.0   $36 = 100.0   $37 = 0   $39 = 1   $40 = 0   $43 = 1   $44 = 4   $45 = 3   $46 = 0   $62 = 0   $63 = 3   $64 = 0   $65 = 0   $100 = 250.000    (X-axis travel resolution, step/mm)$101 = 250.000    (Y-axis travel resolution, step/mm)$102 = 250.000    (Z-axis travel resolution, step/mm)$110 = 500.000    (X-axis maximum rate, mm/min)$111 = 500.000    (Y-axis maximum rate, mm/min)$112 = 500.000    (Z-axis maximum rate, mm/min)$120 = 10.000    (X-axis acceleration, mm/sec^2)$121 = 10.000    (Y-axis acceleration, mm/sec^2)$122 = 10.000    (Z-axis acceleration, mm/sec^2)$130 = 200.000    (X-axis maximum travel, millimeters)$131 = 200.000    (Y-axis maximum travel, millimeters)$132 = 200.000    (Z-axis maximum travel, millimeters)$341 = 0   $342 = 30.0   $343 = 25.0   $344 = 200.0   $345 = 200.0   $346 = 1   $370 = 0   $372 = 0   $384 = 0   $398 = 35   $481 = 0   $484 = 1   ok

Web-design, Woodworking, 3D-Art 

 
 

What is the settings from the command $$?

wyzarddoc, 2023-08-31:
Still getting a "Alarm XYZ Probe Cycle Start " alarm and 
"An error was detected while sending '$X': (error:79) An unknown error has occurred. Streaming has been paused.>>> $X"
if this info helps

Web-design, Woodworking, 3D-Art

breiler, 2023-08-31:
Your input pins needs to be inverted, they seem triggered by default.

I tried deciphering the settings ($4 and $14) from grblHAL documentation here but there is information missing so you need to ask them on their github page:
https://github.com/grblHAL/core/wiki/Additional-or-extended-settings

The error 79 stands for NotAllowedCriticalEvent, probably due to the input pins being triggered:
https://github.com/grblHAL/core/blob/master/errors.h#L108

wyzarddoc, 2023-08-31:
Thanks for your help. On the 1st link you sent their seems to be a mask at $22 using 8 bits but no mention how to set the bits, binary,hex or octal and the nex $ is $28 which doesn't correspond to any binary format.. I'll try inverting the probe and xyz bits again thanks for you help.
Web-design, Woodworking, 3D-Art 

 
 

https://github.com/grblHAL/core/wiki/Additional-or-extended-settings

https://github.com/grblHAL/core/blob/master/errors.h#L108

pogo555, 2023-09-03:
I have this problem. UGS starts, connects reports the conversation then disconnects and will not reconnect.

My engraver is a Vevor 1610 which uses a Vigo Tec GRBL controller.

Following the advice above, I installed the current version (v2.0.20) but the fault remains. I have attached the log file as requested above in the hope that something simple can be found. 

Pogo
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/12505185/messages.log)

breiler, 2023-09-03:
@pogo555 the log doesn't reveal anything other than that UGS could not open the connection to the device. Make sure that no other software is connected to the device and that you have the right user permissions to access `/dev/ttyUSB0`. (https://support.arduino.cc/hc/en-us/articles/360016495679-Fix-port-access-on-Linux)

With some distributions you need to uninstall `brltty` (https://learn.adafruit.com/ladyadas-learn-arduino-lesson-number-0/install-software-linux#driver-installation-1849770)

If that doesn't work, please create a new issue as your problem is not related to grblhal.

pogo555, 2023-09-04:
Thanks for the quick response, @breiler No, brltty was not running.  I'll be away from the bench for a few days. l'll have a poke around and see if I can get some more clues when I get back. 

Pogo

c---, 2023-09-06:
I'm having this same problem with a similar VigoTec controller.
```
Build info:
Grbl 1.1f ['$' for help]
VIGOTEC.SDEM1.1.0.220328

[VER:]
[OPT:V,15,511]
```
It connects and gets the status then immediatlly "disconnects" but it's not really disconnected because it's still polling the controller. Reconnecting doesn't work because "busy".

Here is the Console log:
```
*** Connecting to jserialcomm://ttyUSB0:115200
>>> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
ok
>>> $I
[VER:]
[OPT:V,15,511]
ok
>>> $$
$0 = 8    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
$33 = 0   
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 1000.000    (X-axis maximum rate, mm/min)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 20.000    (Z-axis acceleration, mm/sec^2)
$130 = 300.000    (X-axis maximum travel, millimeters)
$131 = 183.000    (Y-axis maximum travel, millimeters)
$132 = 45.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
```
The `.ugsplatform/var/log/messages.log` has this:
```
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Connecting to GRBL on port ttyUSB0
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.universalgcodesender.communicator.AbstractCommunicator]: Connecting to controller using class: JSerialCommConnection with url jserialcomm://ttyUSB0:115200
SEVERE [UGSEventDispatcher]: Could not dispatch the event ControllerStateEvent to the listener PortAction
SEVERE [null]: Last record repeated again.
```

wyzarddoc, 2023-09-06:
I would try sudo apt-get autoremove brltty  --> this opens up usbtty0 conflicts with other drivers if your on a linux machine.Check Adafruit's website they have some more info good luck.

c---, 2023-09-06:
@wyzarddoc Are you referring to my issue? I'm not using Ubuntu and not running brltty or anything else that has the serial port open (lsof will show if anything has it). Other serial software like Arduino, bCNC, minicom, UGS v2.0.14, etc have no issues.

breiler, 2023-09-06:
@c--- please create a new issue.

c---, 2023-09-06:
@breiler  Sorry, thought this one seemed exactly the same but I guess other issues were present. Mine created as #2309

wyzarddoc, 2023-09-06:
Sorry saw ttyUSB0 and thought Linux

breiler, 2023-09-07:
If you post a description of the alarm condition and its code I might be able to point you in the right direction for help.

The classic GRBL will not run on the ESP32.

breiler, 2024-02-07:
Closing as we did not get further responses from the user.

closed 2024-02-07 by breiler