winder/Universal-G-Code-Sender #2296: jSerialComm drops connection after connectng to GBRL

PopcornAWH, 2023-08-22:
### Version

2.0.18

### Hardware / Firmware

GRBL 1.1

### What happened

When I  connect my Genmitsu 4040-Pro, initialization starts, and right after that the USB connection breaks. Reconnecting isn't possible, no ports found anymore.

### How to reproduce

Hookup a 4040-Pro, and connect.

Probably that specific machine  will not be available. When needed, I can give remote access to investigate.

### Operating System

Windows 10-22H2 64b

### Anything else

Happens every time. 
Came from 2.0.14, there it connects and works, but gives "ALARM HOLD SOFT RESET CYCLE START"

breiler, 2023-08-22:
I made this adjustment to be able to connect to Genmitsu controllers a while back:
https://github.com/winder/Universal-G-Code-Sender/pull/2255

It is available in 2.0.19 or later

breiler, 2023-08-22:
The version is available here: https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.19

PopcornAWH, 2023-08-22:
Oh, hadn't found that, thanks!

19 is the only version I hadn't tried... now wonder why ;-)

---
Paul

Links:
------
[1] https://github.com/winder/Universal-G-Code-Sender/pull/2255
[2] 
https://github.com/winder/Universal-G-Code-Sender/issues/2296#issuecomment-1688310807
[3] 
https://github.com/

breiler, 2023-08-22:
Sorry about that, this version is completely freshly baked and I haven't published it yet.

PopcornAWH, 2023-08-23:
Got a chance to try 2.0.19.  Unfortunately, it does the same thing...
I'll attach the logmessages and comsole output.

Let me know if you need anything else.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/12415098/messages.log)

[Console_Output.txt](https://github.com/winder/Universal-G-Code-Sender/files/12415099/Console_Output.txt)

breiler, 2023-08-23:
This is likely the problem, a new variant of writing the version string which UGS can't understand:
```
>>> $I
[VER:ARM32 V2.6.20230529:]
[OPT:VZL,35,254]
```

Is this their new control board?
[image]

PopcornAWH, 2023-08-23:
Yep That's the board

breiler, 2023-08-23:
Sorry, the issue was auto closed when I merged my fix.

Please try the latest nightly build from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

PopcornAWH, 2023-08-23:
Your fix works... 
[image]

Anything to do about the 'ALARM: HOLD SOFT RESET CYCLE START'?

breiler, 2023-08-23:
Can you right click and turn verbose logging in the console and copy'n'paste the one of the status strings that is reported by the controller?

PopcornAWH, 2023-08-23:
Like this?

<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS|WCO:0.000,0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS|Ov:100,100,100>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS|WCO:0.000,0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS|Ov:100,100,100>

breiler, 2023-08-23:
Yes, thanks!

The controller reports that the pins for these features are triggered `Pn:RHS`:
R: Reset
H: Hold
S: Cycle start

UGS displays them as alarms which maybe is not entirely correct...
But the problem is that this comes from the controller, I can post a question to the Sainsmart support.

(For reference https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports)

PopcornAWH, 2023-08-23:
Thanks... as said, machine functions regardless... So not very urgent.

breiler, 2023-10-23:
I did not get a response from Sainsmart. I did however remove the alarm label, it will now only show the activated pins. 

As this is something that is configured in the firmware or by connecting the pins in a certain way there is not much more we can do from UGS.

I am closing this issue.

closed 2023-10-23 by breiler

PopcornAWH, 2023-10-24:
Pity they didn't react...

Thanks for looking into it...

Can confirm alarm is removed now

---
Paul

Links:
------
[1] 
https://github.com/winder/Universal-G-Code-Sender/issues/2296#issuecomment-1775791901
[2] 
https://github.com/

PR #2297 (2023-08-23): Handle new variant of Genmitsu version string
Fixes #2296