winder/Universal-G-Code-Sender #2299: Opening a previously-opened file doesn't always work when "show editor when opening gcode files" is disabled

bertieconfundo, 2023-08-26:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

I'm using the latest code from github as of Aug 26, 2023.  I've taken the time to root-cause this issue but I'm having difficulty figuring out what the right solution is.

To reproduce, you must have the "Show editor when opening g-code files" option unchecked.

If you open FileA.nc, and then open FileB.nc, and then try to open FileA.nc again, sometimes it won't load FileA.nc and just leave FileB.nc loaded.  This can be dangerous if you're not in the habit of verifying the file actually loaded as the machine won't behave as expected.

The cause of this issue is in the constructor of GcodeDataObject.java.  In the constructor, you're checking the "show editor when opening g-code" preference, and then based on this, you either set the OpenCookie so it can be found by OpenFileAction.java, or you directly call openFile (from EditorUtils.java) which opens the gcode file in the back end only.

The problem is, that this constructor is being called as a side-effect of OpenFileAction.actionPerformed when it calls DataObject.find() for the file you just told it to open.  The first time you open a file, this causes the constructor to run in GcodeDataObject.java, but subsequent times it's not guaranteed to happen if the object hasn't been destroyed.

Complicating this is that you've got different DataObject types (gcode, and UGS designer files), which are both being handled by OpenFileAction.

I think the right solution here is to contain this all within ugs-platform-gcode-editor, and somehow make the GcodeDataObject and SourceMultiviewElement check the showOnOpen setting when the OpenCookie.open() is run (not in the constructor of the GCodeDataObject), and then decide if it should open the editor window, or just load the gcode in the backend without showing the editor window.

### Operating System

MacOS

breiler, 2023-10-23:
I may have fixed this by registering an open cookie when loading the dataobject. I couldn't reliable reproduce this problem on my computer so any help verifying is much appreciated!

breiler, 2024-01-23:
I am closing this as I am no longer able to reproduce it.

closed 2024-01-23 by breiler