winder/Universal-G-Code-Sender #2303: Return to Zero not considering limits when Z-axis reversed

PopcornAWH, 2023-08-31:
### Version

2.0.20

### Hardware / Firmware

GRBL 1.1

### What happened

I might be doing something wrong, but I encountered this:

After Homing, when machine is at X0Y0Z0, if you press return to zero, it causes a limit-alarm because it wants to move to the safety height set in options. The safety height can only be set >1 (default 5).

I know returning to Zero right after homing is an unusual scenario, but consider POS X100Y100Z-4, it still would cause a limit-alarm with default safety height.

Having the safety height is a good feature, but it would be nice if it considered the limits.

Verbose output (excerpt):

Grbl 1.1f ['$' for help]
Grbl for ARM32
Version:ARM32 V2.6
$0 = 10    (step pulse,usec)
$1 = 25    (step idle delay,msec)
$2 = 0    (stepport invert mask)
$3 = 4    (dirport invert mask)
$4 = 0    (stepenable invert,bool)
$5 = 1    (lims pin invert,bool)
$6 = 1    (probe pin invert,bool)

 $H

ok
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS|WCO:0.000,0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS|Ov:100,100,100>
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS>

 G21G90 G0Z1   **_// It wants to go to Z1, but my Z is inverted: $3 = 4    (dirport invert mask)_**
 G90 G0 X0 Y0
 G90 G0 Z0

ok
<Run|WPos:0.000,0.000,0.211,0.000|Bf:33,254|FS:228,0|Pn:RHS>

[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
An unexpected error was detected: (ALARM:3) Reset while in motion. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.
[MSG:Check Limits]
Grbl 1.1f ['$' for help]
Grbl 1.1f ['$' for help]
Grbl for ARM32
Version:ARM32 V2.6

[image]

### How to reproduce

Invert Z-axis, set safety height to 5, move to POS X100Y100Z-4, press Return to Zero.

### Operating System

WIN10 22H2

breiler, 2023-08-31:
Thanks for the feedback.

The first fix might be to make it possible to set safety height to zero - basically to disable it.

To fix this we need to handle another edge case where some controllers work in positive machine coordinate space which puts the machine zero at lower front left corner. It might be solvable due to this https://github.com/winder/Universal-G-Code-Sender/issues/2301#issuecomment-1700385784 but it will require some work.

PopcornAWH, 2023-08-31:
> (comment) [1] but it will require some work.

FYI

I also work in XY positive and Z negative space... I think a lot of CNC 
routing/lasering machines do the same...

Disabling it removes a safety feature, could a relative move be an 
option? It would be necessary to check if it doesn't cross the limits 
though, but being close to zero in negative space could be considered 
safe to move imo...

---
Paul

Links:
------
[1] 
https://github.com/winder/Universal-G-Code-Sender/issues/2301#issuecomment-1700385784
[2] 
https://github.com/winder/Universal-G-Code-Sender/issues/2303#issuecomment-1700801788
[3] 
https://github.com/