winder/Universal-G-Code-Sender #2305: UGS crash....Java crash after an amount of time

itsnotlevel1, 2023-09-02:
### Version

2.0.20

### Hardware / Firmware

FluidNC

### What happened

I have been using UGS on my linux box to control my machine for a short amount of time but have recently switched to my macmini that I moved to the shop. After a while, what seems like a few min to an hour??? UGS crashes ORRR java is crashing. I have yet to have java crash on anything or ugs crash on my linux box. Not sure if its UGS or Java. Just posting here to see if its me or something else. thanks!

### How to reproduce

Run UGS normally and it will eventually crash or its java crashing. 

### Operating System

Ventura 13.5.1 (22G90)

### Anything else

Mac Mini 2023 M2 chip

Not where to find the java log file.

itsnotlevel1, 2023-09-02:
Just got another crash..... Java report:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               java [4851]
Path:                  /Applications/Universal Gcode Platform.app/Contents/PlugIns/jdk-13.0.1+9-jre/Contents/Home/bin/java
Identifier:            java
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        bash [4752]
User ID:               501

Date/Time:             2023-09-02 18:09:00.1696 -0400
OS Version:            macOS 13.5.1 (22G90)
Report Version:        12
Anonymous UUID:        5FB8F362-ACE1-BBE3-4716-CA892FC23A7F

Sleep/Wake UUID:       F5869DF5-668E-407B-B32D-A851AF75C7D0

Time Awake Since Boot: 19000 seconds
Time Since Wake:       10182 seconds

System Integrity Protection: enabled

Crashed Thread:        46  Java: pool-1-thread-1

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x7c719b25caa800a4 -> 0x00001b25caa800a4 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x7c719b25caa800a4

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   java [4851]

VM Region Info: 0x1b25caa800a4 is not in any region.  Bytes after previous region: 29368091410597  Bytes before following region: 75703988518748
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Application Specific Information:
abort() called

Thread 0::  Dispatch queue: com.apple.main-thread
0   ???                           	    0x7ff8994169a8 ???
1   libsystem_kernel.dylib        	    0x7ff8097c85c2 mach_msg2_trap + 10
2   libsystem_kernel.dylib        	    0x7ff8097d673d mach_msg2_internal + 78
3   libsystem_kernel.dylib        	    0x7ff8097cf5f4 mach_msg_overwrite + 692
4   libsystem_kernel.dylib        	    0x7ff8097c88aa mach_msg + 19
5   CoreFoundation                	    0x7ff8098e2a40 __CFRunLoopServiceMachPort + 145
6   CoreFoundation                	    0x7ff8098e14cb __CFRunLoopRun + 1365
7   CoreFoundation                	    0x7ff8098e091c CFRunLoopRunSpecific + 560
8   HIToolbox                     	    0x7ff81382af3d RunCurrentEventLoopInMode + 292
9   HIToolbox                     	    0x7ff81382ad4e ReceiveNextEventCommon + 657
10  HIToolbox                     	    0x7ff81382aaa8 _BlockUntilNextEventMatchingListInModeWithFilter + 64
11  AppKit                        	    0x7ff80c8dc25c _DPSNextEvent + 858
12  AppKit                        	    0x7ff80c8db106 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1214
13  libosxapp.dylib               	       0x12cfd19cd -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 121
14  AppKit                        	    0x7ff80c8cd788 -[NSApplication run] + 586
15  libosxapp.dylib               	       0x12cfd1814 +[NSApplicationAWT runAWTLoopWithApp:] + 222
16  libawt_lwawt.dylib            	       0x12d320223 +[AWTStarter starter:headless:] + 834
17  JavaNativeFoundation          	    0x7ffa257f08ef +[JNFRunLoop _performCopiedBlock:] + 15
18  Foundation                    	    0x7ff80a6ec773 __NSThreadPerformPerform + 177
19  CoreFoundation                	    0x7ff8098e2906 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
20  CoreFoundation                	    0x7ff8098e28a9 __CFRunLoopDoSource0 + 157
21  CoreFoundation                	    0x7ff8098e2686 __CFRunLoopDoSources0 + 217
22  CoreFoundation                	    0x7ff8098e130a __CFRunLoopRun + 916
23  CoreFoundation                	    0x7ff8098e091c CFRunLoopRunSpecific + 560
24  libjli.dylib                  	       0x10c6993fb CreateExecutionEnvironment + 400
[... 1569 lines cut ...]
      "deploymentId" : 900000017
    }
  ]
}
}

Model: Mac14,3, BootROM 8422.141.2, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M2, Apple M2, Built-In
Display: LEN LI2321swA, 1920 x 1080 (1080p FHD - Full High Definition), Main, MirrorOff, Online
Display: LEN LI2321swA, 1920 x 1080 (1080p FHD - Full High Definition), MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Jun 11 2023 06:02:28 version 23.20.95.0.40.50.92 FWID 01-da9eb52a
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB2.0 Hub
USB Device: CP2102N USB to UART Bridge Controller
USB Device: USB-C 4-in-1 Multiport
USB Device: Microsoft® Nano Transceiver v1.0
USB Device: USB Billboard Device
USB Device: USB31Bus
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

itsnotlevel1, 2023-09-02:
[JavaCrashReport.txt](https://github.com/winder/Universal-G-Code-Sender/files/12504802/JavaCrashReport.txt)

breiler, 2023-09-03:
I don't own a Apple M2 device so it is hard for me to debug this. So any help with this would be appreciated.

itsnotlevel1, 2023-09-03:
YES YES! :-) Ill help any way I can....just tell me what you need.

breiler, 2023-09-03:
The first thing I would try is replace the Java runtime with another variant.

Start with downloading the "All platforms" edition of UGS:
https://github.com/winder/Universal-G-Code-Sender#downloads

This does not contain a Java runtime environment, so that needs to be downloaded and unzipped separately from here:
https://adoptium.net/temurin/releases/?os=mac&arch=aarch64&version=11

Then you need to edit the `ugsplatform/etc/ugsplatform.conf` and change the config parameter:
`jdkhome="jdk/jdk-13.0.1+9-jre"` to point to the your JDK directory.

Then start UGS with the script `ugsplatform/bin/ugsplatform`

itsnotlevel1, 2023-09-03:
Thank you I'll try all this tonight!

Paul

-------- Original Message --------

itsnotlevel1, 2023-09-04:
Ok, Not sure but I think I did everything correctly but now im getting a terminal opening up before UGS opens with this: 

Last login: Mon Sep  4 12:01:39 on ttys000
/Users/paulwilson/Downloads/ugsplatform/bin/ugsplatform ; exit;
paulwilson@MINIMACDRAWING ~ % /Users/paulwilson/Downloads/ugsplatform/bin/ugsplatform ; exit;
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/Users/paulwilson/Downloads/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

but this never happened before, and I havent run UGS long enough today for it or java to crash. I am running now so I will post.

breiler, 2023-09-04:
The terminal showing errors are normal when running from the start script. If this works I will try and package a new UGS variant for MacOSX for Aarch64 architecture. Thanks for testing this!

breiler, 2023-09-04:
You are now running Java for an architecture which might not be supported by the library we are using for accessing the gamepad. But as I don't have access to a machine with that architecture I have no way of verifying this. 

Post your log from `~/Library/Application Support/ugsplatform/var/log/messages.log` after your job is done and we can see if it gives any clues.

itsnotlevel1, 2023-09-06:
Most of the day on Monday after your recommendations and about 3 hours tonight with no crashes. I will continue to run the machine and leave UGS running until tomorrow and the same for tomorrow evening after I get off work and will leave UGS open but not connected until then. Just as a point of reference, even if I wasnt using the machine but had ugs open and connected, it would still crash. As of now, it seems to have stopped. 
I am running the java free version of 2.0.20 of UGS and OpenJDK11U-jre_aarch64_mac_hotspot_11.0.20.1_1 version of java, just so you know.

itsnotlevel1, 2023-09-06:
BTW.... there is no such file or folder for UGS....
~/Library/Application Support/ugsplatform/var/log/messages.log

itsnotlevel1, 2023-09-06:
Third day, no crashes....still running. Just an update.

breiler, 2023-09-08:
Thanks, I'll try and create a new packaging for MacOSX aarch64.

itsnotlevel1, 2023-09-25:
Hej! Joacim.... :-) Any progress on a new build for MacOSX? I can help if you would like. Anything I can do I am happy to. Thanks!

itsnotlevel1, 2023-10-06:
Hey again Joacim! I pulled the new Pre-release down and used it for a while today. It crashed twice both times I wasnt using it but I was using it for about 2 hours before that. It seems like it crashes when Im NOT using it.... :( Everything else seems to be working....Im really enjoying having my controller back. Below is the crash info:

[21prereleasemaccrash.pdf](https://github.com/winder/Universal-G-Code-Sender/files/12835618/21prereleasemaccrash.pdf)

itsnotlevel1, 2023-10-26:
So for the last week I have downloaded a nightly build and the program opens then just sits there with my spinning apple balloon and doesnt actually start but its half way open trying...???? to load?

[image]

Video.... https://drive.google.com/file/d/1g4HyJt0IfCMmWyH-CazLMBzMiLFz_X8i/view?usp=drive_link

itsnotlevel1, 2023-10-26:
Maybe some cache I need to delete and start over?

breiler, 2023-10-26:
That version should contain the feature that clears the cache on startup.
Did you download the MacOSX ARM64 package?

If not or if you are unsure, there is a release candidate that you can try here: https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.1.0

breiler, 2023-10-26:
If that still doesn't work I would appreciate the log file from `~/Library/Application Support/ugsplatform/var/log/messages.log`

itsnotlevel1, 2023-10-26:
I deleted some files in macos pertaining to UGS....they were pretty hidden but now seem to be working......I seriously tried to open it up several times in the last few days and it would just hang. I cleared those files.....I can look to see what they were....and now it opens. 
Ill continue to use it as normal....thanks for all your attention and hard work!!

itsnotlevel1, 2023-10-26:
Here are the logs anyway..... and the files I hunted down and deleted. 

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13180991/messages.log)
[Archive.zip](https://github.com/winder/Universal-G-Code-Sender/files/13181007/Archive.zip)

closed 2025-07-19 by itsnotlevel1