winder/Universal-G-Code-Sender #2306: UGS Stops

Pipinval, 2023-09-04:
### Version

2.0.20

### Hardware / Firmware

Other (add details about it below)

### What happened

I recently built a pen plotter, with Arduino Nano and a cnc shield V4.  With Grbl [v1.1h] https://github.com/gnea/grbl/releases

When I press play and the UGS starts, it seems to be working fine, but a while passes and it freezes in the middle of the CNC work. Just before freezing a message appears something like:

[Error] Error mientras se procesaba la respuesta 
<<Idle|MPos:-13.1821,-14.3573,0.0000|FS:0.0,700Ov:0,00,100|A:S>>: For input string: "700Ov:0"

Once stopped, if I press play again a window appears with the message:    "Error al  comenzar el flujo de archivo"

As stated here:  https://winder.github.io/ugs_website/guide/troubleshooting/
I modified Xms to something like -J-Xms256m

I also reinstalled JAva, And I also bought another cable, but nothing. The problem continues

Any idea how to solve this?

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/12515022/messages.log)

### Operating System

Windows 10 x64 .  Intel(R) Core(TM) i7-5500U  2.40GHz   8,00 GB RAM

### Anything else

This always happens, it doesn't matter what code you put in, after a while the program stops, the amount of time is random, it may be just at the beginning or when it has a little left to finish.
This always happens, it doesn't matter what code you put in, after a while the program stops, the amount of time is random, it may be just at the beginning or when it has a little left to finish.

breiler, 2023-09-04:
The garbled data that you showed here is likely caused by interference in the USB connection and can not be fixed in software:
[image]

If your Arduino is a cheaper chinese variant it may use the CH340 serial driver which may be the culprit (https://github.com/grbl/grbl/issues/1825#issuecomment-1482333878).

It might help with a high quality USB cable or properly ground your machine.

Pipinval, 2023-09-04:
> The garbled data that you showed here is likely caused by interference in the USB connection and can not be fixed in software: [image]
> 
> If your Arduino is a cheaper chinese variant it may use the CH340 serial driver which may be the culprit ([grbl/grbl#1825 (comment)](https://github.com/grbl/grbl/issues/1825#issuecomment-1482333878)).
> 
> It might help with a high quality USB cable or properly ground your machine.

Hello Breiler,
Thanks for such a quick response.
It is indeed a cheap Arduino.
I installed the CH341SER.EXE driver, but the error still occurs.
So right now I will order an original arduino nano, if you have any suggestions for another good alternative brand, I would be grateful.

When I have the new arduinoles I will inform you

breiler, 2023-09-04:
Sorry, I don't have any recommendations, maybe someone else here can pitch in?

I am closing this as there isn't much we can do in UGS to fix this.

closed 2023-09-04 by breiler