winder/Universal-G-Code-Sender #2310: Cut type does not change path on imported ellipse

WbrJr, 2023-09-09:
### Version

2.0.20

### Hardware / Firmware

GRBL 1.1

### What happened

Hey there! Thanks for the awesome tool!
I am new to cnc, but i think i found an unwanted feature.
I imported an dxf with two ellipses/circles. When choosing and changing the cut type when one ellipse is selected, the path in the visualizer does not change. when i recreate one ellipse with the internal tool, it works without issue and the path in  the visualizer changes like expected.
I attached a picture, I hope it clarifies what i mean.

[image]

### How to reproduce

I used OnShape to export a face of a part ( just a flat ring, so two circles with the same center point) as a dxf file.
I imported the file to UGS and separated the ellipses from the group and wanted to cut the inside circle on the inside of the path and the outer circle on the outside.
changing the cut type in the cut settings tab does not change the generated path in the visualizer. It seems the cutting path is on the path/curve, which leads to the cut circles being off by the tool diameter.

in the end I just used the included ellipse tool to recreate the circles and there i could change the cut path to inside or outside with the visualizer path changing like expected.

it happened both in the 2.0.20 and 2.0.18 version.

### Operating System

Windows

### Anything else

I hope its not my fault and it is just a setting i overlooked. Thanks anyways!

breiler, 2023-09-09:
Can you attach the DXF for me to debug with

WbrJr, 2023-09-09:
I could not attach it here, dxf are not supported. So here is the link to my drive, it has the ugs and dxf inside. I hope it helps.

https://drive.google.com/drive/folders/1Ene5-6r2i8dA9hWAYSTRSHbkr7rCiabg?usp=drive_link

breiler, 2023-09-10:
Not exactly sure what you mean.

I tried importing the DXF, and made a new circle with the exact dimension and position as the outer circle and it looks like it generates the exact same gcode for both circles:
[image]

Looking closer at your example, there is something that seems slightly off, do they have the same dimensions and/or position?
[image]

Can you save a ugsd-project file that generated the image that you originally posted?

WbrJr, 2023-09-10:
The link above to my drive contains both the original dxf and the ugs file. Yeah, the circles are not quite concentric, but that is not what I meant :) 
I'm not shure how I could help you recreate my issue, other than providing my original files. I don't quite remember all my steps to get there, I will try later to do it again, maybe I will notice when it goes wrong.

breiler, 2023-09-13:
We'll close it for now, if you encounter this again and have some more information we can just reopen it.

closed 2023-09-13 by breiler