winder/Universal-G-Code-Sender #2311: NullPointerException when trying to open NC file V 2.0.20 GRBL 1.1

BWalker7788, 2023-09-10:
### Version

2.0.20

### Hardware / Firmware

GRBL 1.1

### What happened

Trying to open an project file getting the following error...

java.lang.NullPointerException
	at org.netbeans.modules.editor.fold.ui.CodeFoldingSideBar.resolveBackColor(CodeFoldingSideBar.java:353)
	at org.netbeans.modules.editor.fold.ui.CodeFoldingSideBar.updateColors(CodeFoldingSideBar.java:368)
	at org.netbeans.modules.editor.fold.ui.CodeFoldingSideBar.access$200(CodeFoldingSideBar.java:99)
	at org.netbeans.modules.editor.fold.ui.CodeFoldingSideBar$2.preferenceChange(CodeFoldingSideBar.java:231)
	at org.netbeans.modules.editor.fold.ui.CodeFoldingSideBar.addNotify(CodeFoldingSideBar.java:287)
	at java.desktop/java.awt.Container.addNotify(Unknown Source)
	at java.desktop/javax.swing.JComponent.addNotify(Unknown Source)
	at org.netbeans.modules.editor.impl.CustomizableSideBar$WestSidebarHolder.addNotify(CustomizableSideBar.java:288)
	at java.desktop/java.awt.Container.addNotify(Unknown Source)
	at java.desktop/javax.swing.JComponent.addNotify(Unknown Source)
	at java.desktop/java.awt.Container.addNotify(Unknown Source)
	at java.desktop/javax.swing.JComponent.addNotify(Unknown Source)
	at java.desktop/java.awt.Container.addNotify(Unknown Source)
	at java.desktop/javax.swing.JComponent.addNotify(Unknown Source)
	at java.desktop/java.awt.Container.addImpl(Unknown Source)
	at java.desktop/java.awt.Container.add(Unknown Source)
	at org.openide.text.CloneableEditorInitializer.initCustomEditor(CloneableEditorInitializer.java:523)
	at org.openide.text.CloneableEditorInitializer.initCustomEditorAndDecorationsInEDT(CloneableEditorInitializer.java:573)
[catch] at org.openide.text.CloneableEditorInitializer.run(CloneableEditorInitializer.java:321)
	at org.openide.text.CloneableEditorInitializer.processPendingEDTRequests(CloneableEditorInitializer.java:135)
	at org.openide.text.CloneableEditorInitializer.waitForFinishedInitialization(CloneableEditorInitializer.java:103)
	at org.openide.text.CloneableEditor.getEditorPane(CloneableEditor.java:673)
	at org.netbeans.core.spi.multiview.text.MultiViewEditorElement.getEditorPane(MultiViewEditorElement.java:146)
	at com.willwinder.ugs.nbp.editor.SourceMultiviewElement.componentActivated(SourceMultiviewElement.java:81)
	at org.netbeans.core.multiview.MultiViewPeer.peerComponentActivated(MultiViewPeer.java:353)
	at org.netbeans.core.multiview.MultiViewCloneableTopComponent.componentActivated(MultiViewCloneableTopComponent.java:124)
	at org.openide.windows.WindowManager.activateComponent(WindowManager.java:266)
	at org.netbeans.core.windows.WindowManagerImpl.lambda$notifyRegistryTopComponentActivated$1(WindowManagerImpl.java:1204)
	at org.netbeans.modules.openide.util.NbMutexEventProvider$Event.doEvent(NbMutexEventProvider.java:97)
	at org.netbeans.modules.openide.util.NbMutexEventProvider$Event.readAccess(NbMutexEventProvider.java:72)
	at org.netbeans.modules.openide.util.LazyMutexImplementation.readAccess(LazyMutexImplementation.java:66)
	at org.openide.util.Mutex.readAccess(Mutex.java:241)
	at org.netbeans.core.windows.WindowManagerImpl.notifyRegistryTopComponentActivated(WindowManagerImpl.java:1203)
	at org.netbeans.core.windows.Central.activateModeTopComponent(Central.java:1786)
	at org.netbeans.core.windows.WindowManagerImpl.topComponentRequestActive(WindowManagerImpl.java:1368)
	at org.openide.windows.TopComponent.requestActive(TopComponent.java:836)
	at org.openide.windows.CloneableOpenSupport$1.run(CloneableOpenSupport.java:83)
	at org.netbeans.modules.openide.util.NbMutexEventProvider$Event.doEvent(NbMutexEventProvider.java:97)
	at org.netbeans.modules.openide.util.NbMutexEventProvider$Event.writeAccess(NbMutexEventProvider.java:62)
	at org.netbeans.modules.openide.util.LazyMutexImplementation.writeAccess(LazyMutexImplementation.java:56)
	at org.openide.util.Mutex.writeAccess(Mutex.java:291)
	at org.openide.windows.CloneableOpenSupport.open(CloneableOpenSupport.java:79)
	at org.openide.text.CloneableEditorSupport.open(CloneableEditorSupport.java:412)
	at com.willwinder.ugs.nbp.core.actions.OpenFileAction.actionPerformed(OpenFileAction.java:61)
	at com.willwinder.ugs.nbp.core.actions.OpenAction.openFile(OpenAction.java:104)
	at com.willwinder.ugs.nbp.core.actions.OpenAction.actionPerformed(OpenAction.java:98)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:136)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

### How to reproduce

Open any file in v2.0.20 

### Operating System

Windows 11 pro and Rasp os

### Anything else

This is happening on the windows x64 2.0.20 release and All platforms.

BWalker7788, 2023-09-10:
@winder help

BWalker7788, 2023-09-10:
Not sure what the difference is but with the nightly build (Version2.0-SNAPSHOT / 2023-09-05) files are loading in perfectly.

breiler, 2023-09-10:
Can you post your log files from:
C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log

BWalker7788, 2023-09-10:
[Logs.zip](https://github.com/winder/Universal-G-Code-Sender/files/12570717/Logs.zip)

breiler, 2023-09-11:
I have no idea why this is happening. 

You can try closing UGS and then delete the cache directory here:
`C:\Users\Rusty\AppData\Local\ugsplatform\Cache\var\cache`

Then start again and see if that fixes the problem.

BWalker7788, 2023-09-11:
I deleted the cache and it was still happening. I was able to get it working by clearing the cache again and then uninstalling  and reinstalling though. @breiler Thank you for help.

closed 2023-09-11 by BWalker7788