winder/Universal-G-Code-Sender #2317: Hard limit has been triggered

lidershinobi, 2023-09-12:
### Version

2.0.20

### Hardware / Firmware

GRBL 1.1

### What happened

The problem is when the cnc starts cutting the plane, when it goes halfway it throws this error:
An error was detected while sending 'G01X20.3257Y-14.2476': (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended. Streaming has been paused.

### How to reproduce

Well, my 350x350x160mm printer is large and I have the endstops in -x-y+z.

When I open the program, first a homing and a Zero reset, I place a 10x12 pcb board in the center so I have a macro to move it between X100X100 and -100X since the home z is at the top. So I move manually and place the position on the corner plate by touching it and again Reset Zero, with or without autolevel I start. I open the gcode file made in flatcam.

and I start the cnc to mill the plate. When it is halfway through, the error appears:
An error was detected while sending 'G01X20.3257Y-14.2476': (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended. Streaming has been paused.

I had done the same tests before and it was not giving an error, and I tried to reinstall several versions and the problem still exists. Yes, once the cnc was starting, I put my hand on the tweezers with which I do the autolevel and I threw out that error.

It is as if it were touching the xy limits or the probe clamps are detecting some signal, or there is some limit problem in the flatcam design or in the gbrl program.

### Operating System

windows 10 64bit

breiler, 2023-09-12:
This sounds like a classic EMI problem which is likely caused by a high powered spindle or dust collector. It is a hard problem to solve and can not be solved in software.

A short time solution can be to deactivate hard limits.

Read about electrical noise here:
https://www.cnccookbook.com/cnc-electrical-noise-grounding-techniques-interference-filter/

lidershinobi, 2023-09-13:
I understand. What I did not mention is that I have the 2 positive and negative clamps for the probe connected to an Arduino shield v4 nano board in the negative and SCL connection, what I am sure of is that this version of the board is poorly designed.
I will try connecting the clamps to the min z connectors and see what happens, since the SCL input is for communication and it may be detecting noise.

lidershinobi, 2023-09-13:
ok I just solved the problem, what caused the endstops to activate was because the 12v motor cable was together with the endstop cables.

This caused that when the motor was turned off, it generated an electromagnetic wave that generated a small voltage in the endstops, which the board detected.

For new users who are building a homemade cnc, move the cables and relays that activate and deactivate the 12v cnc motor or another away from the stop cables or the probe cables.

Thank you Breiler for your support and comment that helped me solve it.

closed 2023-09-13 by breiler