winder/Universal-G-Code-Sender #2318: Add bridges/supports

sergey-das, 2023-09-12:
### Description

Some of Gcode generators, especially focused on PCB milling provide great feature: add bridges/supports for cutout operations.
For example I have some plate (plywood, fiberglass etc) and I want to cutout a contour. The material is fixed not using double sided stick tape, but by some clamps. So when the contour is close to be completed the middle part can go off the position and break the whole cutout process. If we add some thin bridges to the contour, the middle part fill be always perfectly fixed and aligned and then after milling/cutting process is finished it will be easy to cutout it using some kind of wire cutter.

breiler, 2023-09-12:
In other words, you want to add a tabs feature to the designer tool?

breiler, 2026-08-28:
Fixed in #3181

It is available in the latest nightly and will become available in 2.1.27

closed 2026-08-28 by breiler