winder/Universal-G-Code-Sender #2319: UGS Setup Wizard cannot recognize the controller

rcecinati, 2023-09-14:
### Version

2.0.20

### Hardware / Firmware

GRBL 1.1

### What happened

Hello, some features of UGS appear unreachable to me, first of all the SetupWizard, claiming it can't recognize  my controller, and forcing me to a manual setup. I performed the  $100 and $101 settings, and they are consistent with the behavior of my machine so far.
Jogging works great
"Set zero" and "Return to zero" functions give no result. The only way to reset the coordinates is disconnecting/reconnecting.
"Visualize" function does not work too

### How to reproduce

Hardware: Arduino UNO compatible (AZ-Delivery)
CNC Shield 3
2 x Stepper Motor Driver
Tried UGS 2.0.18 and 2.0.20 
I am running GRBL 0.9j because the version  1.1 seems to be incompatible with Shield 3 (Joacim's warning), it actually   sticks down the motors
The drivers were trimmed about the load current, and work fine
I am using only 2 axis X+Y, and corrected the homing sequence manually in config file  excluding homing on Z axis

### Operating System

Windows 11 64bit

### Anything else

It always happens , trying to run the wizard. I have experienced a success in the very first stages of my development, couldn't say what broke the magic.
Further detail are of course available, e.g. the Firmware Settings

breiler, 2023-09-14:
To get the Setup Wizard working you need to use the latest nightly build. I added a fix for older versions of GRBL a [couple of days ago](https://github.com/winder/Universal-G-Code-Sender/pull/2314). Download it from here: https://github.com/winder/Universal-G-Code-Sender#downloads

> "Set zero" and "Return to zero" functions give no result. The only way to reset the coordinates is disconnecting/reconnecting. "Visualize" function does not work too

What is your $10 setting?
Make sure you have a reporting level that reports the machine position and/or work positions. 
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#10---status-report-maskbinary

For instance `$10=3`

rcecinati, 2023-09-14:
Dear Joachim,
thank you for your lightning responses.
I have tried the nightly build, and it works as regards the Setup Wizard.
Good job.

About the Set zero and Return to zero commands, it seems they are not
affected by correction of the $10 parameter to the value 3 (formerly 1).
Tomorrow I will dive further

Thank you

Riccardo

breiler, 2023-10-15:
Closing as this is now available in 2.0.21

closed 2023-10-15 by breiler

notusenn, 2024-11-21:
I got the same issue, and downloaded nightly build, still same, how to fix?
I didn't find a way to uninstall the UGS2.1.9, so maybe the two Interact with each other？
thanks

breiler, 2024-11-21:
@notusenn, please create a new issue and provide more details. What does the console log say  after you've tried connecting to the controller? What controller and firmware do you have?