winder/Universal-G-Code-Sender #2327: Support for grbl mega 5x

siboj, 2023-10-02:
### Description

Hi,

As i see UGS can work with grbl mega 5x, but problem is when you homing axis its only do for 3 axis. It would be nice to add full support for grbl mega 5x. I didn't test to much grbl mega 5x on UGS so i don't know what else is missing.

breiler, 2023-10-03:
The homing is configured in the firmware. To add more axes you need to change the controller config, recompile and upload.
 
Have a look here: https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/config.h#L241

closed 2023-10-03 by breiler

siboj, 2023-10-09:
> The homing is configured in the firmware. To add more axes you need to change the controller config, recompile and upload.
> 
> Have a look here: https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/config.h#L241

What exactly to change?
My output is 
[image]

breiler, 2023-10-10:
That is not the output from homing. You are pressing "Return to zero", which moves the machine to the current work position zero. I do not want to extend this button to move additional axes, you will have to create your own macro button for that.