winder/Universal-G-Code-Sender #2328: Visualizer not operational with v2.0.20

billyxmoto, 2023-10-03:
### Version

2.0.20

### Hardware / Firmware

GRBL 1.1

### What happened

gcode visualization not showing with v2.0.20, seems OKI with v 2.0.18

### How to reproduce

loaded usgplatform64.exe from version 2.0.20.  opened gcode file, no visualization
loaded usgplatform64.exe from version 2.0.18.  opened same gcode file,  visualization seems OK
see screen shots 

[image]
[image]

### Operating System

WIN 10Pro 22H2 Thinkpad-T410 Intel(R) Core(TM) i5 CPU   

### Anything else

Consistent

breiler, 2023-10-04:
Start the 2.0.20 version and then exit it. After that attach the messages.log file from:
`C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log`

breiler, 2023-11-17:
Closing as we didn't get a response

closed 2023-11-17 by breiler