winder/Universal-G-Code-Sender #2332: Cant load G-Code

Paulmac1966, 2023-10-07:
### Version

2.0.20

### Hardware / Firmware

GRBL 1.1

### What happened

I have just upgraded to the latest version and now cant load G-Code, it flashes a preview up on the visualiser briefly then disappears and no G-Code has been loaded

### Operating System

Product Version: Universal Gcode Platform 20230825 Java: 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9 Runtime: OpenJDK Runtime Environment 13.0.1+9 System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform) User directory: C:\Users\User\AppData\Roaming\ugsplatform Cache directory: C:\Users\User\AppData\Local\ugsplatform\Cache\var\cache

breiler, 2023-10-07:
Can you try and close UGS then delete the folder `C:\Users\User\AppData\Local\ugsplatform\Cache\` and then try and start again.

If that doesn't work I'm interested in your log file at:
`C:/Users/User/AppData/Roaming/ugsplatform/var/log/messages.log`

Paulmac1966, 2023-10-08:
Removing the cache directory worked

Thank you

breiler, 2023-10-15:
Many users are experiencing this, I have never been able to reproduce it.
I have added an option to clear the cache on startup which hopefully will fix this problem for future users.

closed 2023-10-15 by breiler

Paulmac1966, 2023-10-15:
Ok, thanks for the update.